ros-controls/ros_control #201: Jade Release

rctoris, 2015-04-07:
Just a friendly ping for a potential jade release to help some blocking packages for the upcoming beta.

rethink-imcmahon, 2015-04-14:
+1 for this

adolfo-rt, 2015-04-15:
I'll be spending time on the project in the next weeks, and will take a look at what's needed here. Because of the ROS distro transition, I'd like to take the opportunity to review which behavior and API changes would make sense to incorporate. More concretely, I'm thinking at least of #75 (API break), and https://github.com/ros-controls/ros_controllers/issues/48 (behavior change), which is long overdue.

davetcoleman, 2015-04-15:
Lets add https://github.com/ros-controls/ros_controllers/pull/142 also

scpeters, 2015-04-22:
+1, as [ros_control blocked the gazebo_ros_pkgs release for indigo](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/179)

adolfo-rt, 2015-04-23:
Its likely that I'll have some cycles next week to look into this.

wjwwood, 2015-04-23:
+1, let me know if I can help.

adolfo-rt, 2015-04-28:
If `gazebo_ros_control` is moved to the `ros-controls` organization, (https://github.com/ros-simulation/gazebo_ros_pkgs/issues/179), then it would no longer block the release of `desktop-full`, and of Jade's beta.

Usually, the release of `ros_control` and friends has lagged a bit from the core packages. For indigo, we did not make it to the beta, but we did make it to the official release date. I'm trying to aim for a similar timeline with Jade.

adolfo-rt, 2015-05-26:
Quick update. My progress on this has been preempted until mid-June because of higher priority tasks. Most (if not all) of the development effort involving API changes has been done in #204, but I still have some TODOs before commiting to the release.

DLu, 2015-07-02:
Not to pile on, but +1 for new features of urdf_tutorial.

bduffany, 2015-08-29:
Any updates on this?

adolfo-rt, 2015-09-04:
Hi,
The deadline I've set to myself is ROScon. I'll be offline for three weeks starting on Tuesday Sept 8.

I'd like to fix #211, and optionally consider moving `gazebo_ros_control` to the ros-controls organization, since it's currently living in a limbo state. I'm actively working on that package, so I hope to have something solid for ROScon.

Thanks all for your prolonged patience. The jade release brings quite a few changes that have taken time to get right.

werner-ne, 2015-09-28:
Hi Adolfo, Is your personal deadline until end of this week still valid? thx

adolfo-rt, 2015-10-01:
Hi. I'm waiting for feedback on #211, which I consider a blocker for the Jade release.

TheDash, 2015-11-03:
Can we release this into Jade? It is blocking https://github.com/pr2-gbp/pr2_controllers-release/issues/1

davetcoleman, 2015-11-04:
+1, can we also fix the velocity controller's feed-forward term?

adolfo-rt, 2015-11-13:
All blocking PRs for releasing the `ros_control` repo have been submitted (#218 and #210). After a pair of eyes gives them a +1, I'll merge and proceed with the pre-release → release sequence.

The idea is to get `ros_control` and `ros_controllers` released first. Then I'll see when I find another slot for `gazebo_ros_control`, which needs to be migrated to Gazebo 5. I don't know if there's a high overhead for this task or not. Hopefully not.

> +1, can we also fix the velocity controller's feed-forward term?
ros-controls/ros_controllers#142

I commented directly in #142. It could make it as well.

scpeters, 2015-11-13:
Regarding `gazebo_ros_control` and `gazebo5`, there's two concerns:

* gazebo5 uses c++11, so anything downstream that builds plugins needs to use the `GAZEBO_CXX_FLAGS`, but that should already be fixed ( https://github.com/ros-simulation/gazebo_ros_pkgs/commit/4d46186a2fa20bdc59403bfb34271a672021361b )
* We changed the behavior of `Joint::SetVelocity` and `Joint::SetMaxForce`, so anything using velocity commands (instead of torque commands) will need to use `Joint::SetParam("vel")` and `Joint::SetParam("max_force")`. I'm not sure if `gazebo_ros_control` uses a velocity control interface.

adolfo-rt, 2015-11-18:
Thanks @scpeters. The second point has not yet been addressed, and `gazebo_ros_control` does use a velocity control interface.

adolfo-rt, 2015-11-20:
FYI,
https://github.com/ros/rosdistro/pull/9849
https://github.com/ros/rosdistro/pull/9850

Pending feature requests can be merged as patch releases on top of 0.10

adolfo-rt, 2015-11-21:
I'll push some release notes to the robot control SIG in the coming days.

davetcoleman, 2015-11-22:
Thanks @adolfo-rt !

ipa-mdl, 2015-12-01:
Now that the jade release is on its way, what about a new indigo release?

bmagyar, 2015-12-02:
+1
The `diff_drive_controller` could definitely use one!

I'm willing to do the release procedure once all lights are green.

adolfo-rt, 2015-12-04:
Yes, on the agenda. I'm currently getting a failed test in `diff_drive_controller`. Can't release `ros_controllers` like that:

```
[diff_drive_controller.rosunit-diff_drive_limits_test/testAngularJerkLimits][FAILURE]
/home/adolfo/ros_control_ws/src/ros_controllers/diff_drive_controller/test/diff_drive_limits_test.cpp:150
The difference between new_odom.twist.twist.angular.z and 0.7 is 0.060075881205914228, which exceeds JERK_ANGULAR_VELOCITY_TOLERANCE, where
new_odom.twist.twist.angular.z evaluates to 0.76007588120591418,
0.7 evaluates to 0.69999999999999996, and
JERK_ANGULAR_VELOCITY_TOLERANCE evaluates to 0.050000000000000003.
```

I don't have time right now to dig into this. So if someone can reproduce that would accelerate the release.

scpeters, 2015-12-04:
I'll try to take a look. It may be related to the first point I made above.

efernandez, 2015-12-06:
That test from the `diff_drive_controller` is probably not robust enough to time differences when it computes the relative odometry: it works for me (I can't reproduce it) and it works on Travis.

Anyway, I'm going to look into it to make it more robust to time differences. If that's not the reason, I'll increase the tolerance on the test for now, as this feature (jerk limits) shouldn't be blocking a release.

adolfo-rt, 2016-02-12:
Indigo:

https://github.com/ros/rosdistro/pull/10491
https://github.com/ros/rosdistro/pull/10492

I still see that diff drive test flakiness in my box, but pulled the trigger anyway.

efernandez, 2016-02-12:
@adolfo-rt I have some WIP branch for them, but I can relax that test that sometimes fails meanwhile.

In my branch (I repeat still WIP) I have an exhaustive set of unit tests that all pass, but curiously, that rostest doesn't, so I think it's related with some timing issue between runs.

adolfo-rt, 2016-02-15:
@efernandez Let's bring in those changes in the next patch release. This one is already out.
https://github.com/ros/rosdistro/pull/10497
https://github.com/ros/rosdistro/pull/10498

closed 2016-02-15 by adolfo-rt

sytelus, 2016-07-23:
Looks like this issue is closed but there is still no ros-jade-gazebo-ros-control package :(.

`sudo apt-get install ros-jade-gazebo-ros-control`
`E: Unable to locate package ros-jade-gazebo-ros-control`

bmagyar, 2016-07-25:
Are you sure everything is ok with your installation?
https://github.com/ros/rosdistro/blob/master/jade/distribution.yaml#L1104
I checked packages.ros.org but couldn't find it there... it's weird. 
Could you bring up this question here please? http://discourse.ros.org/c/release
I don't have an installation to test this on at the moment.

wjwwood, 2016-07-25:
Jade hasn't been synced with Public since `gazebo_ros_pkgs` was released with `gazebo_ros_control` included: http://repositories.ros.org/status_page/ros_jade_default.html?q=gazebo_ros_control

bjornite, 2016-07-25:
Does that mean this issue will go away soon?

wjwwood, 2016-07-25:
It means that the package will be available next time we sync Jade to the public repositories. I don't know when that will be, but probably within the next month.

You could use the shadow fixed repository and get the package now if you wanted: http://wiki.ros.org/ShadowRepository