ros-controls/ros_control #208: [rqt_controller_manager] Jade compatibility, UI revamp

adolfo-rt, 2015-08-21:
This issue discusses the `rqt_controller_manager` specifics contained in #204. I'd like to present the most important changes and discus them here before merging #204.

Apart from making the rqt plugin compatible with the ROS API changes that are coming to jade, I took the opportunity to improve certain usability aspects. A detailed description can be found in the commit log of 0f1f50e67e6757799cebe386ff5069eec5d1ca26. Here, I'll discuss them in more graphical terms.

**More compact UI**
I decided to provide the same information as before, but in a different way. The plugin now looks like this:

[image]

which is an important improvement in terms of horizontal screen real-estate. Also, you can see that uninitialized controllers are shown together with the loaded and running ones.

**Detailed controller info**
The above image shows the bare minimum needed to monitor controllers. If a user wants to see more, she can *double-click* a specific controller, and a pop-up will display the additional information. Notice in the following image how controllers claiming resources from multiple intefaces are properly supported

[image]

**Read-only main plugin area**
In the same spirit as before, users need to call a context menu to step the lifecycle of a controller, so no possibly dangerous actions can be taken by accident. One difference is that this now also applies to uninitialized controllers, which before you could load and start from the main widget. Below is an example of the context menu accessible with a *right-click*.

[image]

A special ping goes to @kphawkins, the original plugin author, who got me into writing rqt plugins.

ipa-mdl, 2015-08-21:
Great! I really like the clean, compact interface :)

adolfo-rt, 2015-08-25:
@kphawkins, apologies for not waiting enough to get your feedback. Considering the dates, you might be on holidays. My allocated time for performing `ros_control` maintenance is coming to an end, and I'm merging all outstanding PRs.

I'll merge #204, and if there are any outstanding issues, we can address them directly on `jade-devel`.

One thing I'd like to point out is that your original implementation had no copyright statement. I added [this line](https://github.com/ros-controls/ros_control/blob/1d16edf65b4a5eb4672fc424eaac6de2e835fd9d/rqt_controller_manager/src/rqt_controller_manager/controller_manager.py#L2),  as IIUC Georgia Tech is the affiliation under which you wrote the code. Let me know if it needs to be differently.

closed 2015-08-25 by adolfo-rt