ros-controls/ros_control #213: Issues/Questions on joint_limits_interface

davetcoleman, 2015-10-23:
I reviewed the joint_limits_interface.h file this week and have some thoughts that I could turn into PRs if desired:

I found the variables initialized within the realtime loop [throughout](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L206) the code. Isn't this always bad?

Probably a dumb question, but why is there a min effort but not a [min velocity](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L307)? Velocity is absolute but effort is not?

Both the EffortJointSaturationHandle and the VelocityJointSaturationHandle throw exceptions when they are missing limits, but the PositionJointSaturationHandle does not. This is misleading - the Position handle should also fail [instead
of silently disabling itself](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L80)

Again, I'm happy to create a PR if its a good idea!

adolfo-rt, 2015-11-13:
> I found the variables initialized within the realtime loop [throughout](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L206) the code. Isn't this always bad?

What is undesired are heap allocations in the real-time path. The line you point to is creating a double instance on the stack. Heap allocations (`new`, `malloc`) are in general non-deterministic, stack allocations are, as it's basically a pointer offset on the (already allocated) stack memory region.

> Probably a dumb question, but why is there a min effort but not a [min velocity](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L307)? Velocity is absolute but effort is not?

There isn't any notion of min effort, other than [this](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L286), which is also how min velocity is handled.

> Both the EffortJointSaturationHandle and the VelocityJointSaturationHandle throw exceptions when they are missing limits, but the PositionJointSaturationHandle does not. This is misleading - the Position handle should also fail [instead
of silently disabling itself](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L80)

[EffortJointSaturationHandle](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L266) does not throw, but [VelocityJointSaturationHandle](https://github.com/ros-controls/ros_control/blob/indigo-devel/joint_limits_interface/include/joint_limits_interface/joint_limits_interface.h#L421) does. I agree that there should be a consistent behavior. In particular, if no limitation is taking place at all for lack of specified parameters, it would make sense to raise an exception.

davetcoleman, 2015-11-18:
Thanks for these useful responses @adolfo-rt 

When I get a chance I'll submit a PR for making sure an exception is properly thrown

davetcoleman, 2015-12-04:
PR created: https://github.com/ros-controls/ros_control/pull/221

closed 2018-02-01 by bmagyar