ros-controls/ros_control #219: Adding new maintainers

adolfo-rt, 2015-11-20:
I would be interested in having other people share the load of merging PRs and making releases. There are currently two persons, @efernandez and @bmagyar that have offered to help with some of these tasks.

Because I'm a member and not an owner of this organization, I cannot grant them member access. So, could one of the owners please take action? (add me as an organization owner and/or add them as members).

Pinging current owners: @ahendrix, @davetcoleman, @tfoote, @wjwwood.

TIA

wjwwood, 2015-11-20:
@adolfo-rt I promoted you to a team manager, which should allow you to add people to the `ros_control` team on the organization. Let me know if that is sufficient for you to add the user's in question.

bmagyar, 2015-11-25:
Thanks for the support!

efernandez, 2015-11-30:
Thanks!

@adolfo-rt Let us know when you make us members.

adolfo-rt, 2015-12-01:
Only organization owners can send invitations for others to join. That part at least I still cannot do.

wjwwood, 2015-12-01:
@adolfo-rt can you check to make sure you cannot invite them to the `ros_control` team? This note in the GitHub interface leads me believe that you should be able to since I made you a team maintainer:

[image]

adolfo-rt, 2015-12-02:
[A team maintainer](https://help.github.com/articles/giving-team-maintainer-permissions-to-an-organization-member/) *should* be able to add/remove members, but when I try to [add a new member](https://help.github.com/articles/adding-organization-members-to-a-team/), I get the following:

[image]

See how users are grayed-out, and the message stating `Only organization owners can Send Invitations`. No clue why I'm getting this.

wjwwood, 2015-12-02:
Weird. I guess you can add/remove people to the team but not the organization? Either way I just sent them invitations.

efernandez, 2015-12-02:
Thanks @adolfo-rt @wjwwood

bmagyar, 2015-12-02:
Thanks guys!

closed 2015-12-02 by bmagyar

adolfo-rt, 2015-12-03:
Much appreciated @wjwwood