ros-controls/ros_control #223: Resources list order changed in `controller_manager/list_controllers`

scpeters, 2015-12-08:
I've been trying to make a tool for printing out the upper and lower limits of joints used in a specific controller. I recently tried calling the `controller_manager/list_controllers` service since that prints a list of controllers and their joints under the resources field. I've run into the problem that this list becomes disordered (alphabetized) somehow.

Consider the hand controllers on robonaut for example, which have an [unalphabetized list of joints](https://bitbucket.org/nasa_ros_pkg/nasa_r2_common/src/40d7cea220cb245ff727e8c08ce7dbfbdb47d7d9/r2_moveit_config/config/controllers.yaml?at=indigo-devel#controllers.yaml-26:41) that starts with:

~~~
      - r2/left_arm/hand/index/yaw
      - r2/left_arm/hand/index/proximal
      - r2/left_arm/hand/index/medial
~~~

This order is preserved when I echo the controller state:

~~~
$ rostopic echo -n 1 /r2/l_hand_controller/state | grep joint_names 
joint_names: [
'r2/left_arm/hand/index/yaw',
'r2/left_arm/hand/index/proximal',
'r2/left_arm/hand/index/medial',
'r2/left_arm/hand/ringlittle/little', 'r2/left_arm/hand/middle/yaw', 'r2/left_arm/hand/middle/proximal', 'r2/left_arm/hand/middle/medial', 'r2/left_arm/hand/ringlittle/ring', 'r2/left_arm/hand/thumb/roll', 'r2/left_arm/hand/thumb/proximal', 'r2/left_arm/hand/thumb/medial', 'r2/left_arm/hand/thumb/distal']
~~~

However, when I call the `controller_manager/list_controllers` service it changes the order:

~~~
$ rosservice call /r2/controller_manager/list_controllers
...
  - 
    name: l_hand_controller
    state: running
    type: effort_controllers/JointTrajectoryController
    hardware_interface: hardware_interface::EffortJointInterface
    resources: [
'r2/left_arm/hand/index/medial',
'r2/left_arm/hand/index/proximal',
'r2/left_arm/hand/index/yaw',
'r2/left_arm/hand/middle/medial', 'r2/left_arm/hand/middle/proximal', 'r2/left_arm/hand/middle/yaw', 'r2/left_arm/hand/ringlittle/little', 'r2/left_arm/hand/ringlittle/ring', 'r2/left_arm/hand/thumb/distal', 'r2/left_arm/hand/thumb/medial', 'r2/left_arm/hand/thumb/proximal', 'r2/left_arm/hand/thumb/roll']
  -
~~~

I'm using indigo. Is there a sort being called somewhere in the service stack? I peeked at [controller_manager.cpp](https://github.com/ros-controls/ros_control/blob/indigo-devel/controller_manager/src/controller_manager.cpp#L602-L638) and I didn't see a sort called.

scpeters, 2015-12-08:
I just realized that `controller_interface` [uses a `std::set` for controller resources](https://github.com/ros-controls/ros_control/blob/indigo-devel/controller_interface/include/controller_interface/controller_base.h#L144). So this is fundamental to the API, not a small thing to fix.

davetcoleman, 2015-12-08:
> I've been trying to make a tool for printing out the upper and lower limits of joints used in a specific controller

Won't most ros_control implementations simply read the joint limits from the robot's URDF or from the parameter server? Could your tool just read the joint limits from their source?

scpeters, 2015-12-08:
I'm reading the joint limits from the URDF, but I need an ordered list of the joint names to create the vector of joint limits (motivated by trying to modify the [r2_hand.py script](https://github.com/osrf/rosbook/blob/master/chessbot/r2_hand.py#L11-L15) from the ROS book but having trouble because the joint names and order have changed). I thought the `list_controllers` service would be a nice way to get this ordered list, but `std::set` causes the order to be lost.