ros-controls/ros_control #227: I am trying to create my own controller but it won't load with error: "Could not create object of class type"

cyborg-x1, 2015-12-14:
It is currently very similar to the imu controller, I also created my own interface. Everything compiles the library is there but I get this error message when I try to load it:

Could not load class h4r_ev3_sensor_control/Ev3UltraSonicRangeController: MultiLibraryClassLoader: Could not create object of class type h4r_ev3_sensor_control::Ev3UltraSonicRangeController as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1450114480.594795807]: Could not load controller 'Ev3UltraSonic' because controller type 'h4r_ev3_sensor_control/Ev3UltraSonicRangeController' does not exist.

Does this sound familiar to you somehow?
Any idea, what I am doing wrong?

My code is here:
https://github.com/Hacks4ROS/h4r_ev3_ctrl/tree/develop/h4r_ev3_sensor_control

Currently I am just trying to get some value out so there is no real interface code there just some value dummies. The library file exists inside my workspace. Also the workspace works, I can launch the node having the hardware interface in h4r_ev3_ctrl.

See include and source folders. In the config there is the test.yaml and in launch the launchfile I load the controller. Also the controller appears in the list of available controllers when I call the service.

How do I make sure that it is "explicitly loaded through MultiLibraryClassLoader::loadLibrary()" ?

cyborg-x1, 2015-12-14:
Awww...

Searching the whole day for this, asking a question here ... seeing it some minutes after .. :(

Forgot this one in the controllers source:

PLUGINLIB_EXPORT_CLASS(h4r_ev3_sensor_control::Ev3UltraSonicRangeController, controller_interface::ControllerBase)

closed 2015-12-14 by cyborg-x1

bmagyar, 2015-12-14:
Doing
`rosservice call /controller_manager/list_controller_types`
shows all the visible controllers for your controller manager (for it's pluginlib loader to be more specific). It usually reveals declaration & export xml issues early on.

cyborg-x1, 2015-12-15:
Well it was in there despite missing this line in the source, but that makes completely sense, the plugin file (that is probably used for this lookup) was in there and correct. I tried around and with the plugin file being wrong the application loading controllers would not even start, directly saying that the macro is wrong.Well, I just overread the PLUGINLIB_EXPORT_CLASS macro when reading sources, that was also when I tried my first controller class which did not workout because of the realtime publishers msg variable being blocked with "boost::noncopy". I tried to create a variable interface for different message types with a template where the whole message object would be needed to be copied to it, but I guess that is not supported because of the realtime/non realtime aspect.