ros-controls/ros_control #232: Controller Manager reload-libraries

mllofriu, 2016-03-16:
I am running an node with an instance of the RobotHW class that creates a controller manager instance, as described in https://github.com/ros-controls/ros_control/wiki/hardware_interface. 

After the node is executing, I perform the sequence:
 - rosrun controller_manager controller_manager load <somecontroller>
 - change and recompile controller
 - rosrun controller_manager controller_manager reload-libraries
 - rosrun controller_manager spawner <somecontroller>

I expected the controller manager to load the new implementation, but it keeps loading the older version. Only when I restart the node it loads the new implementation.

mkoval, 2016-07-07:
I am seeing the same behavior. Did you figure out an alternative way of debugging a controller without restarting the controller manager after every change?

mllofriu, 2016-07-08:
Nope. Restart the controller manager everytime so far.

asimay, 2016-07-08:
what's your load launch file?

mkoval, 2016-07-12:
> what's your load launch file?

I'm not sure what you mean. I have been doing development by loading my hardware interface (a C++ program that instantiates a controller manager) in one terminal window. Then, in a second terminal window, I use `spawner` to load the controller that I am testing. It's currently simple enough that no launchfile is involved.

> Nope. Restart the controller manager every time so far.

Bummer. I was hoping I was doing something wrong. 😦

bmagyar, 2016-07-13:
Hi guys,

Not sure what to say about this. This issue all boils down to [this line](https://github.com/ros-controls/ros_control/blob/kinetic-devel/controller_manager/include/controller_manager/controller_loader.h#L71), while the intention with the reload-libraries functionality was exactly to allow on-line debugging without having the need to restart things AFAIK.

`lsof` may give clues as to which files are open (as to see shared libraries loaded)
It may be worth looking at what happens to some simple code that only uses Pluginlib the same way.

mllofriu, 2016-07-14:
It's been a while since I had this problem. I seem to recall that I think that that reset did not took care of it. I think (because its hard to follow code through these three complex projects) that the underlying class_loader keeps track of loaded libraries on the background: https://github.com/ros/class_loader/blob/indigo-devel/src/multi_library_class_loader.cpp. Maybe calling shutdownAllClassLoaders or unloadLibrary one by one?