ros-controls/ros_control #233: Use a unique NaN to indicate "inert" state

cottsay, 2016-03-28:
I'd like to propose a convention for indicating that a particular actuator or joint go into an "inert" state.

We have found this would be useful in both of our use cases involving `ros_control`. On our servo, we need a way to "relax" the servo. Also on our motor controller, which actuates the main driving wheels, sending a velocity of "0" actually engages braking when at speed. We'd like to have a way to tell the motor controller to "coast".

Both of these functions are available at the hardware interface, but the control framework only allows the controllers to communicate their commands via "double" values.

Many folks don't know that there are quite a few possible values with represent a NaN in a floating point number. Some of these are "signaling" NaNs, and some are "quiet" NaNs. I think we should define a special "quiet" NaN value for representing this state (and possibly other states).

If a controller sent the special NaN to a hardware interface which wasn't expecting it, it would be interpreted as any other NaN would. Inversely, if we use anything but the default quiet NaN value, a controller shouldn't ever unwittingly tell the hardware interface to enter the inert state.

If we exclude the default quiet NaN value, we have 21 bits to play with.

The only code change here would be to define the value somewhere with a macro, and to write a macro "function" for detecting if a given float is the special value (similar to `isnan(x)`). If someone could tell me where in the headers this would be appropriate, and what value would be best, I'd be happy to open a PR. We should also settle on the right term (ie inert, free, released, relaxed, etc).

Looking forward to thoughts on this.

Thanks!

--scott

bmagyar, 2016-03-30:
Hi Scott,

The VelocityJointInterface receives target velocities which is then made available to your hardware abstraction level for further processing before passing to other layers (if any) and eventually to motor drivers.

I don't think that at the level of this framework one should not be concerned about what 0 means for your controller. A target velocity of 0 means to stop.

Producing a control signal that is smooth in velocity profiles is the job of the controller you are running. 

Correct me please but it seems that you already have your own controller and your own hardware abstraction layer using the current ros_control framework, right? 

Our policy is to leave platform-dependent information out, unless it is generalizable into some control concept.

To think of something conceptually clean that implements inert states I would use 2 interfaces for these joints, one of position or velocity, while the other could be effort. Though effort is often used for torque control, in this case it could be used as a "0 or non-0" flag. A 0 on the effort interface would make it clear that the motor should do absolutely nothing, disregarding other commands. This could for instance be used with actuators where you can set maximum effort along with position or velocity targets (eg: some dynamixel motors).

bmagyar, 2018-02-01:
I'm closing this issue now, if there's anything you wish to add please reopen.

closed 2018-02-01 by bmagyar