ros-controls/ros_control #234: Kinetic release

bmagyar, 2016-04-17:
Calling all stations!

I'd like to push this through as fast as it's possible so we are not blocking the release of `ors-kinetic-desktop-full` (we do through Gazebo).

Are there any PR-s that you'd wish to see merged & volunteer to put effort into it? 
I can do the release rain-dance but I need to make sure that we are on the same page.

As there have already been changes in messages on `control_msgs` I'm going to use kinetic-devel for the new releases. 

Please put your (re)marks below: @adolfo-rt  @davetcoleman @efernandez @ipa-mdl @paulbovbel  @ros-controls/ros_control

efernandez, 2016-04-18:
I can help looking into `realtime_tools`. There are only two things that need attention there:

1. One PR (https://github.com/ros-controls/realtime_tools/pull/8) related with running the tests on Travis, which are currently broken. **This should go into kinetic**, but we have to fix the PR branch before merging it.
2. One issue (https://github.com/ros-controls/realtime_tools/issues/14) related with the lock-free implementation that currently doesn't support a `writeFromRT` method. **I don't see this in kinetic**, as it'd need some development. Apparently, Boost lockfree doesn't support multiple writers, and it's not clear to me that other lock free libraries do.

bmagyar, 2016-04-18:
:+1: @efernandez could you please have a look at the travis stuff for all other `ros-controls` repos? I'm not very fluent there.

jacquelinekay, 2016-04-22:
How is progress on the Kinetic release?

bmagyar, 2016-04-22:
Waiting for some API changes on `control_toolbox`. Once done there, that +
`ros_control` are ready to roll out.

toliver, 2016-05-06:
I would like for these 2 PRs to be merged into Kinetic. I'm not sure how far they are to be considered mergeable, but I can definitely work on fixing any issues.

Both of them are focused on allowing to combine several RobotHW into a single one, and therefore allow a single controller to access resources from more than one RobotHW.

- #220 
- #231 This one builds upon the previous one.

bmagyar, 2016-05-06:
I have released to Kinetic a few days back.
Merging those 2 PRs have been on the table for a while now... could you please close those 2 PR-s and open a new one against kinetic-devel? I think a single PR containing those 2 commits is ok, easier to discuss.

bmagyar, 2016-05-06:
...since the initial kinetic release is done, I'm closing this issue.

closed 2016-05-06 by bmagyar

toliver, 2016-05-06:
I will do that.

We are also interested in https://github.com/ros-controls/ros_controllers/pull/185.

It was also left unmerged. Do you think it can make its way into Kinetic?

bmagyar, 2016-05-06:
Absolutely. I only need you to implement the changes Adolfo asked for there and help me out by providing as many tests as you can think of ;) 
The same rule applies to it though, please close that PR and open a new one after rebasing against kinetic-devel.

toliver, 2016-05-06:
Thanks @bmagyar . I think those changes were already implemented, but we can resume that discussion once we open the new PR.