ros-controls/ros_control #236: Pull request template to all repos of ros-controls

bmagyar, 2016-05-11:
Hello folks,

I gave a shot to pull request templates this morning and I think it would help to have them on the ros-controls repos. 
Github blog entry: https://github.com/blog/2111-issue-and-pull-request-templates

My test repo:
https://github.com/bmagyar/pr-template-test

This is what I drafted up for the test repo, we can start from this:
```
This is a pull request template, read it, then you are free to delete it

Please make sure your PR follows the guidelines below:
* The PR title summarises the problem it solves
* You are adding the minimal amount of commits. (You rebased your changes on the latest of this branch, no merge commits etc)
* If it changes behaviour, tests have been added or adjusted, and they all pass.
* Provided explanation where necessary.
```

Please provide your feedback @ros-controls/ros_control

davetcoleman, 2016-05-21:
I like the idea of a template, but the guidelines you are proposing we add to ros_control seems pretty generic and obvious. Perhaps point to how to run said tests? Are there any other details that might help new users in testing ros_control?

bmagyar, 2016-05-22:
I agree with them being generic and kinda obvious, but we are still receiving many that don't even meet these requirements. Adding help to run the tests is a good idea! 

What do you think @efernandez ?

efernandez, 2016-05-25:
I think it'd help to mention how to run specific tests, since running all of them takes like 1 hour for all. Other than that, it looks fine to me.