ros-controls/ros_control #237: CombinedRobotHW documentation

bmagyar, 2016-05-24:
With the advent of this new feature our lack of documentation & examples increases even more in contrast to the features we have.

I would like to ask @toliver is you could provide some very minimal but working examples (can be taken from the tests for instance) with a little bit of discussion, perhaps a figure showing how the structure of a RobotHW and example configurations to set up. The information in #231 can be a good starting point. Can you do this please?

It could go to [the ROS wiki](wiki.ros.org/combined_robot_hw) and it would be nice to have it in the repo as a .md too.

paulbovbel, 2016-10-27:
Just going to leave this here, since it's probably the best available documentation at the moment :)

https://vimeo.com/187696094

bmagyar, 2016-10-27:
Thanks Paul, I'll update the wiki with this and the older talk of Adolfo
too!

ptiza-v-nebe, 2018-10-01:
Sorry for opening this again, but is there a chance that someone has written a complete example of using combinedhw?

My main question is what relation has combinedhw with some robothw?
Does I have to start first robothw package then combinedhw or does combinedhw start robothw itseld? Where I have then to put the config of robothw? Why I have than to use pluginlib for exporting my robothw? 

There are plenty questions open but no complete examples. I want just a simple scenario. Some 2 dump robothw with some hardcoded states and command, but where I can clearly see that my both robothws are now one.

If I have two robothw packages from two devices or vendors does I have to take it apart or I can use it as is?

For some reason the combined_robot_hw_tests does not build the necessary nodes for running the test.

Thank you!

ptiza-v-nebe, 2018-10-09:
I have managed now to understand how things works. Here some first trying to make some tutorial for the whole picture of how implementing CombinedRobotHW.
https://github.com/ros-controls/ros_control/wiki/Writing-CombinedRobotHW

toliver, 2018-10-25:
@ptiza-v-nebe Thanks for making the effort of writing documentation for this package (something that I didn't get around to do yet).

Just for completion, the `combined_robot_hw_tests` package does in fact contain a running example with 4 different RobotHW instances being loaded by the CombinedRobotHW instance that runs in the node called `dummy_app` (see [here](https://github.com/ros-controls/ros_control/blob/melodic-devel/combined_robot_hw_tests/test/cm_test.test#L66)). Of course that is not a replacement for good documentation :)

ptiza-v-nebe, 2018-10-27:
Ahh yes the tests :) Problem on them was I didn't got them to run. The other side is they are too cryptically for someone who is entering ros-control and maybe ros overall. My wish was to have something globally that I cant just download and start, then to build on and how you said good docu :))
Would it better to move this one to wiki.ros.org? Maybe you can mention some more details on functionality of combohw and I can better this little docu?