ros-controls/ros_control #242: Question Controller Manager: How to know which controller is running on which joint?

barbalberto, 2016-06-13:
Hi guys,

I watched the Adolfo's talk at ROSCon (https://vimeo.com/107507546) and I found it really helpful. 
I managed to implement my hw_interface and move my robot using rqt by following this tutorial https://github.com/ros-controls/ros_control/wiki/hardware_interface

Now, the custom message I use to control my robot will slightly change depending on the controller used to move the joint (position, torque ...).

So, let's suppose my controller_manager is running, all controllers are loaded but none of them is started yet. 

Maybe it is a dumb question, but following the proposed scheme:
~~~
  while (true)
  {
     robot.read();
     cm.update(robot.get_time(), robot.get_period());
     robot.write();
     sleep();
  }
~~~
How do I get to know that no controllers are running yet? Or better, which controllers is running if any?
In case of no controllers running I'd skip the write() for example.

Also, let suppose my robot has 4 joints and, at a certain point in time, I have the following set up:
~~~
joint_1   position controlled
joint_2   velocity controlled
joint_3   torque controlled
joint_4   idle  (no controller is handling this joint)
~~~

I'd like to create the message to sent to my robot in such a way that:

-  It'll hold only references for the first three joints 
- No reference for the joint_4 since no controller is acting on it
- Furthermore while writing my msg I'd like to know that the reference for joint_1 is a position, for joint_2 is a velocity and so on.

What's the best practise to achieve this goal?

Thanks

bmagyar, 2016-06-13:
Hi Alberto,
That loop is flawless. What I would suggest is to smarten up your
RobotHardware and add some mechanism to write() that handles your needs.
One naive way would be to check if the command has changed and only forward
updates to your hardware.

How do you send messages to your hardware? This is where you could encode
the type of the command you are sending. This is - again - all up to you.
On systems where the hardware also talks ROS, I see sensor_msgs/JointState
used often since it is able to carry typed commands.

Handling hardware state switches can be done in the doSwitch() function:
https://github.com/ros-controls/ros_control/blob/kinetic-devel/hardware_interface/include/hardware_interface/robot_hw.h#L147

For instance when a joint can accept both velocity and position commands
depending what you set the hardware to. This is the same spot where you
could add state handling if a joint goes Idle (no controller takes
ownership of it) then it could get an "Idle" flag, so you wouldn't forward
messages to it, or others would get other flags and so you'd know how and
what to forward.

Hope this makes sense,
Bence

barbalberto, 2016-06-15:
> What I would suggest is to smarten up your
RobotHardware and add some mechanism to write() that handles your needs.

Yes, that's exactly the point of the question

> How do you send messages to your hardware? This is where you could encode
the type of the command you are sending.

I use a ROS message and that's what I'm doing. 

I'll do a different example to better clarify my doubt.
Let's I have the following yaml config file for the controller_manager
~~~
controller_a:
    type: position_controllers/JointTrajectoryController
    publish_rate: 50
    joints:
      - joint_1
      - joint_2
    ...

  controller_b:
    type: effort_controllers/JointTrajectoryController
    publish_rate: 50
      joints:
      - joint_1
      - joint_2
    ...
~~~

My managers starts with `controller_a` loaded and running. After a while, `controller_a` is stopped and `controller_b` is started and operating on the same 2 joints. Supposing the switch operation goes ok, my question is: while writing the message, how do I detect this change?

I was thinking to do something like:
~~~
cm.getControllerNames();
cm.getControllerByName();
~~~    
And then ask info to each loaded controller, but the first function is private which suggests me this is not the right approach. So my question is how to collect information about the running controller from the code. There is no way to ask to the controller_manager what is it doing?

> Handling hardware state switches can be done in the `doSwitch()` function

So if I correctly understand what you suggest, I should implement the `doSwitch()` and "take note" of which controller is running and which joint it'll take ownership of, am I right?
If this is the case, shall I use the `prepareSwitch()` or `doSwith()` function?

ipa-mdl, 2016-06-15:
> So my question is how to collect information about the running controller from the code. 

There wasn't any option in the past (except for hijacking `checkForConflict`), so we have added  `prepareSwitch` and `doSwitch`.

> So if I correctly understand what you suggest, I should implement the doSwitch() and "take note" of which controller is running and which joint it'll take ownership of, am I right?

Exactly.

> If this is the case, shall I use the prepareSwitch() or doSwith() function?

`preprareSwitch` runs in non-RT part of the switch sevice handler. `doSwitch` runs in the RT part of `update`.
If you can swtch modes fast enough, just use `doSwitch`.
If the switch can fail for some hardware-specific reason you should check the state within `prepareSwitch` and return false if hardware won't be able to switch (e.g. emergency stop etc.).

barbalberto, 2016-06-22:
Thanks guys, I managed to do it using the  `preprareSwitch` function and it works.
Maybe I'll post the code here for feedbacks later on, but I think I can close the issue now.

Thanks a lot

closed 2016-06-22 by barbalberto

bmagyar, 2016-06-22:
Great! Please if you publish the code let us know so we can reference it.