ros-controls/ros_control #243: RRBoT tutorial controller spawner problem

slaniel, 2016-06-16:
Hi,

I try to control a robot in gazebo with ROS topics.

To learn, I tried the ros_control package for the RRBoT example. 

I can launch the :
`$ roslaunch rrbot_gazebo rrbot_world.launch`
Gazebo opens perfectly with the robot and the environment.

My problem comes afterwards when I execute :
`$ roslaunch rrbot_control rrbot_control.launch`

Everytime my console displays :

```
[INFO] [WallTime: 1466101921.611973] [0.000000] Controller Spawner: Waiting for service controller_manager/load_controller
[WARN] [WallTime: 1466101951.737910] [1599.057000] Controller Spawner couldn't find the expected controller_manager ROS interface.
[rrbot/controller_spawner-1] process has finished cleanly
```

What am I doing wrong?
Sebastien

miguelprada, 2016-06-17:
Hi @slaniel,

First of all, you should specify where those `rrbot_gazebo` and `rrbot_control` packages you're using come from. I'm going to go ahead and assume your talking about the packages in [`gazebo_ros_demos`](https://github.com/ros-simulation/gazebo_ros_demos).

Second, I've just tested the same commands on ROS Indigo with a clean clone of `gazebo_ros_demos` `master` branch and everything works fine. Could you ellaborate more on your problem? Which ROS distro are you using? Have you modified anything in the `gazebo_ros_demos` packages?

Finally, if my assumption is right, maybe you should post your issue in that [repo](https://github.com/ros-simulation/gazebo_ros_demos), not here, since the problem seems more related with incorrect configuration of `rrbot_control`, which is making the controller spawner incapable of finding the right controller manager topics.

adnanmunawar, 2016-07-21:
If you are using Indigo and a version of Gazebo that doesn't come by default with it, this issue arises. A quick fix is to remove the ros-indigo-gazebo7-ros-pgs and install them from source.