ros-controls/ros_control #244: [question] On the FourBarLinkageTransmission type

carlosjoserg, 2016-07-27:
Hi all,

Has anyone ever used this transmission type?

I've been reading for a couple of days, but still can't find a match between the phrase ["This transmission relates two actuators and two joints through a mechanism in which the state of the first joint only depends on the first actuator, while the second joint depends on both actuators......"](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/include/transmission_interface/four_bar_linkage_transmission.h#L47) and a 4-bar linkage, and it bugs me, so I come here for feedback.

My reasoning is that typically one have either a planar or a spatial-spherical (all revolute axes meet at a one point) or a Bennett 4-bar linkage that moves, all of them having 1 d.o.f., and the generic spatial one is over-constrained. So, first thing, I'm not getting why the two actuators.

I understand the remote actuation example, where the two actuators are able to move independently, and the described behavior is achievable. But in a 4-bar linkage using two actuators, the two must move in coordination to follow the constrained configuration space, otherwise the linkage breaks, whence the description does not match to me, and consequently the implemented expression... but only if we still think of a 4-bar linkage.

Thus, my only conclusion so far is that the implementation is actually a _planar_ 5-bar linkage (2 d.o.f.), and then all would be ok, 2 independent actuators, and the state of one joint is determined by 1 actuator, and the other(s) joint state(s) by both actuators (and the inverse map as well).

Any comment, or point that I could be missing?

bmagyar, 2016-08-02:
Hi Carlos,

Let me state first that I have no proper education in mechanics and can only provide an explanation based on intuition and experience.
This transmission was used in the REEM robot, specifically to relate the tilting torso joint to the shoulder tilt axes. To lower the center of mass, the actuators driving to the first shoulder joint (tilt axis) were placed on the lower part of the torso and are connected by belts (there may be a proper name for them :)). This is simple so far, but the complexity is added when you take the tilting of the torso into account, since when that moves it forces the belts to move as well, making the shoulder actuators have to compensate for this. 

You can give it a shot by installing the REEM simulation, I believe it works on Indigo but it's only available from source.
https://github.com/pal-robotics/reem_robot
https://github.com/pal-robotics/reem_gazebo

I made a terrible attempt at drawing it too. So when the red movement is applied, the blue one needs to be there to compensate.
[image]

carlosjoserg, 2016-08-02:
Thanks for the reply @bmagyar ! I tell you, you draw better than me ;)

That's the remote actuation example, where, yes, the shoulder actuator must compensate for the torso actuator motion, if the shoulder relative position w.r.t. ground or torso is to be kept. But moving the shoulder motor does not move the torso or make the torso motor to compensate for nothing, right?

Just found something called the [pantograph](https://en.wikipedia.org/wiki/Pantograph) (which pretty much looks like the mechanism in the left of [this figure](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/images/four_bar_linkage_transmission.png)?), and it is sometimes referred  to as a 4-bar linkage with 2 d.o.f., with a particular set of lengths to be able to scale and replicate motion at the tip. However, this is tricky, the ground is not being counted, that is actually a 5-bar linkage with 2 d.o.f. if you do the count correctly... which leads me to what I think is the answer why this is called FourBarLinkageTransmission, because the ground does not count.

In other words, what is widely-known as a [4-bar linkage](https://en.wikipedia.org/wiki/Four-bar_linkage) with 1 d.o.f., would be a 3-bar linkage within this name encode, because that's 3 bars connected with revolute joints, and the ground does not count, and where I found the mismatch in first place.

Do you agree? 

If you do agree, would you be willing to change the name to either SimpleRemoteTransmission or FiveBarLinkageTransmission? I think it'd be more appropriate (I'd prepare the PR for kinetic, of course)

bmagyar, 2016-08-05:
I'm afraid I cannot make any judgement on this one.
@adolfo-rt , could you please verify this? 

If renaming is strongly supported, we should do it ASAP in Kinetic.

carlosjoserg, 2018-03-16:
I'm gonna close this, I can live with that naming ;) 

Just wanted to be sure that the implementation was one thing or the other, and I think it is clear. 

Thanks!

closed 2018-03-16 by carlosjoserg