ros-controls/ros_control #245: [Multiple controllers] Using ros_control in robot IMAU

dario-pedro, 2016-08-01:
Hi everyone,

My robot (IMAU) has 3 controllers:
```
 - imau_elevator_controller
 - imau_right_gripper_controller
 - imau_left_gripper_controller
```

Each controller contains the joints information, something like:
```
imau_elevator_controller:
  type: position_controllers/JointTrajectoryController
  topic: "test"
  joints:
     - elevator_base_joint
     - elevator_joint
     - prismatic_1_joint
     - prismatic_2_joint
     - gripper_pan_joint
  gains:
    elevator_base_joint: {p: 10000.0, i: 700.0, d: 0.0, i_clamp: 0.0, i_clamp_min: 0.0, i_clamp_max: 0.0}
    elevator_joint: {p: 100000.0, i: 700.0, d: 0.0, i_clamp: 0.0, i_clamp_min: 0.0, i_clamp_max: 0.0}
    prismatic_1_joint: {p: 1000.0, i: 70.0, d: 0.0, i_clamp: 0.0, i_clamp_min: 0.0, i_clamp_max: 0.0}
    prismatic_2_joint: {p: 1000.0, i: 70.0, d: 0.0, i_clamp: 0.0, i_clamp_min: 0.0, i_clamp_max: 0.0}
    gripper_pan_joint: {p: 100.0, i: 0.0, d: 0.0, i_clamp: 0.0, i_clamp_min: 0.0, i_clamp_max: 0.0}
  constraints:
      goal_time: 5555.6
      stopped_velocity_tolerance: 5555.5
      elevator_base_joint: {trajectory: 5555.1, goal: 5555.1}
      elevator_joint: {trajectory: 5555.1, goal: 5555.1}
      prismatic_1_joint: {trajectory: 5555.1, goal: 5555.1}
      prismatic_2_joint: {trajectory: 5555.1, goal: 5555.1}
      gripper_pan_joint: {trajectory: 5555.1, goal: 5555.1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  25
  action_monitor_rate: 10
```

I've managed to load the controllers correctly and everything looks fine on rqt. But when I Start the controllers on rqt, only the last one that was loaded ( imau_left_gripper_controller) seems to be working. By working I mean that the joint_command change when I move the sliders on rqt.

The code can be found  [here](https://github.com/dario-pedro/imau), it was based on [baxter control](https://github.com/davetcoleman/baxter_cpp) by @davetcoleman .

Thanks in advance.

bmagyar, 2016-08-01:
Could you please specify what you mean by "seems to be working"?

If it's rqt, screenshots could work but maybe more direct is if you'd call
Rosservice call /controller_manager/list_controllers

When you think it should all be up and running?
What is your full startup sequence?

facmarques, 2016-08-01:
Hi,

I'm working with Dário and maybe I can explain further.
Ẃe wanted to have 3 (objective is to be generic as possible ) controllers launched from one node.
So we based ourselves on the Baxter Code by @davetcoleman, as baxter has also two arms.

Our sequence is as follows:

Launch robot description;
Controllers spawner;
Node;

The only controller that responds and generates commands it the last one to run inside our code.
If we have one controller that one responds, if we have two the second and so on. 
As you can see in the screenshots below. Only the left gripper has values in the command message (our output for the motors).

Thanks

It seems odd has we stepped into 
[image]
[image]

miguelprada, 2016-08-02:
Since all controllers seem to be running, could you check (e.g. by using `rostopic echo`) that the rqt plugin is really publishing the commands on every `/controller_manager/CONTROLLER_NAME/command` topic (should only publish for the currently selected controller in the widget, I think)?

Also, since you seem to be using the position interface-based joint trajectory controller, no PIDs are used and therefore the gain parameters don't need to be set.

facmarques, 2016-08-02:
Hi,

Yes, we've checked that already (you are correct it only publishes one at a time). But only the  last controller that gets loaded responds.
We were testing several hypotheses for controlling, but thanks we will check the config files to remove te gains.

miguelprada, 2016-08-02:
There should be no reason the controller manager only updates the last controller loaded, just tested this myself with two joint trajectory controllers acting on different joints of the same robot and both respond just fine.

Is the source code for your `RobotHW` implementation available somewhere? It is possible that the issue is lying there somewhere.

bmagyar, 2016-08-02:
The rqt_joint_trajectory plugin only publishes to the currently active controller, and only on change I believe. If you don't touch it, no commands will be send. 

I'd recommend looking at the controller state outputs. Just simply 
`rostopic echo /imau_elevator_controller/state`
The error field here may give you hints as to what reference the controller is trying to track.

@miguelprada , this should be the part where those commands could get stuck IMHO, but it doesn't seem to be the case:
https://github.com/dario-pedro/imau/blob/master/src/Arm_interface.cpp#L93

dario-pedro, 2016-08-02:
Hello,

@bmagyar  I've done` rostopic echo /imau_elevator_controller/state` , and when I move the sliders on rqt plugin nothing changes. The result is showed in the screenshot below:
[image]

@miguelprada , the `RobotHW` code can be found [here](https://github.com/dario-pedro/imau/blob/master/src/Arm_interface.cpp) , if u need any more information , please let me know.

dario-pedro, 2016-08-05:
Hello,
We've tried to sort out the problem using rqt_graph, but we don't understand exactly what's happening. 

The prints below represent the result when you use the sliders of the different controllers.

[image]
[image]
[image]

We're going to try to investigate further into the ros_control code and try to understand what's happening.

bmagyar, 2016-08-08:
Ok, so it seems there's no subscription to those topics. Just to double-check this, please untick "group actions" and "hide leaf topics" "hide dead sinks".

Please also run "rosservice call /controller_manager/list_controllers" and paste the output here.

dario-pedro, 2016-08-09:
Hi again, 
> Ok, so it seems there's no subscription to those topics. Just to double-check this 

The only topic we subscribed is the actuator/motor_nanotec/joint_states on `arm_interface_sub = n.subscribe("/", 1000,&Arm_interface::handleAMotorResponse, this);` to get the robot current states on the `Arm_interface` . The topics we're publish are /actuator/motor_nanotech/"+controller_name+"/joint_command , 1 per controller , done in `joints_pub = nh.advertise<sensor_msgs::JointState>(/*/imau*/"/actuator/motor_nanotech/"+controller_name+"/joint_command", 1000);` .

The prints you requested are posted bellow: ( one for the right_arm and another for the left_arm )

[image]
[image]

The screenshots looks similar , but only the imau_left_gripper_controller seems to be outputting commands.

The information from "rosservice call /controller_manager/list_controllers" is:
```
controller: 
  - 
    name: joint_state_controller
    state: running
    type: joint_state_controller/JointStateController
    hardware_interface: hardware_interface::JointStateInterface
    resources: []
  - 
    name: imau_elevator_controller
    state: running
    type: position_controllers/JointTrajectoryController
    hardware_interface: hardware_interface::PositionJointInterface
    resources: ['elevator_base_joint', 'elevator_joint', 'gripper_pan_joint', 'prismatic_1_joint', 'prismatic_2_joint']
  - 
    name: imau_right_gripper_controller
    state: running
    type: position_controllers/JointTrajectoryController
    hardware_interface: hardware_interface::PositionJointInterface
    resources: ['gripper_right_joint', 'gripper_support_right_joint']
  - 
    name: imau_left_gripper_controller
    state: running
    type: position_controllers/JointTrajectoryController
    hardware_interface: hardware_interface::PositionJointInterface
    resources: ['gripper_left_joint', 'gripper_support_left_joint']

```

bmagyar, 2016-08-09:
Ok, for the sake of completeness, please switch on the top to "Nodes/Topics (all)", your first screenshots were not showing the subscriptions because only active communication was displayed.

dario-pedro, 2016-08-09:
Ok sorry. Here :

[image]

bmagyar, 2016-08-09:
Ok this looks clean and subscriptions are ok so you commands actually make it to the controllers but the commands seem to be ignored. From this point on it's down to debugging your RobotHW.

Give a shot to `rqt_console` and change log levels there to see if you can get some meaningful info out there. You can even compile `joint_trajectory_controller` from source and print stuff there to see where your pipeline is broken (won't be *in* the controller). 
Using multiple controllers is something that's the core of `ros_control`, so the only likely place where things go wrong is your RobotHW code though I couldn't spot any obvious mistakes there.

Perhaps it'd be best to try reimplementing it? This is rather simple stuff so it shouldn't be a big effort. An example for simple `RobotHW` implementation is  [jackal_hardware](https://github.com/jackal/jackal_robot/blob/indigo-devel/jackal_base/include/jackal_base/jackal_hardware.h).

bmagyar, 2018-02-01:
Any update on this? 
I'm going to close this issue bu tfeel free to reopen if you have more news to tell or fixes to submit!

closed 2018-02-01 by bmagyar