ros-controls/ros_control #246: [ROS_CONTROL position_controllers] Tolerances and velocity

facmarques, 2016-08-04:
Hi everyone,

We are currently using position_controllers/JointTrajectoryController.
And we are trying to understand what is the effect of the tolerances and constraints.
We didn't find any documentation that totally explains it.

[image]

Our current behaviour is depicted above.
But we wanted to change the velocity of the motor and damping when arriving at its goal.

[image]

Is it normal that when we remove the feedback the controller keeps on moving the motor ? (see screenshot above)

Another issue that we've run into is that when we populate the velocity fields (in the read method) the controller starts immediately moving to a position without any previous command (the position is somewhat related it the velocity value).
 
Are we supposed to fill the velocity field when working with position_controllers?

Thanks

P.S. We are happy to aid on documenting these issues with the results of this discussion

bmagyar, 2016-08-04:
Hi,
I'll try to group the answers.

1. If I remember well, constraints and tolerances are not used for enforcing anything, but they serve as criterion used for determining whether the executed action was a success or not (reflected in the action result). Read [this](http://wiki.ros.org/joint_trajectory_controller#Parameters)  for more.
2. The way the `joint_trajectory_controller` works is that it generates a trajectory for each joint that is smooth in position, velocity and acceleration profiles. This is achieved by using a Quintic spline interpolation. Both of your images show me curved graphs, hence the generated trajectory is in fact converging _slower_ the closer it gets to the goal. Perhaps the PID-s on your actuators are very aggressive? That may cause jerky motion.
3. What do you exactly mean by removing feedback? If there's no feedback, how do you expect the controller to get there? You can monitor the /CONTROLLER_NAME/state topic to see how the error decreases (or not, when you remove the feedback I guess :))
4. Populating the velocity fields in which read method? Your `RobotHW`? Did you change the source of the `joint_trajectory_controller`? 
5. An older documentation suggests that when filling the velocity fields in the command, the controller would run a quadratic interpolation instead of a quintic one, which may explain the different behaviour, but I'm unsure about how relevant that is to the current controller. You are only expected to fill the position fields, velocity is not needed (have to check the code if the quintic/quadratic switch is still there).

facmarques, 2016-08-05:
**1 - If I remember well, constraints and tolerances are not used for enforcing anything, but they serve as criterion used for determining whether the executed action was a success or not (reflected in the action result). Read this for more.**

Ok, we will check it again. 

**2 - The way the joint_trajectory_controller works is that it generates a trajectory for each joint that is smooth in position, velocity and acceleration profiles. This is achieved by using a Quintic spline interpolation. Both of your images show me curved graphs, hence the generated trajectory is in fact converging slower the closer it gets to the goal. Perhaps the PID-s on your actuators are very aggressive? That may cause jerky motion.**

Yes, the controller is working. We just want it to be more aggressive. Currently, the `joint_trajectory_controller` is only sending commands to the motor in order for the motor to reach 1/10 of its maximum speed. 
The motor's PID are working fine the movement is smooth during all trajectory. The motor is reaching the position sent by the  `joint_trajectory_controller` almost instantly.

**3- What do you exactly mean by removing feedback? If there's no feedback, how do you expect the controller to get there? You can monitor the /CONTROLLER_NAME/state topic to see how the error decreases (or not, when you remove the feedback I guess :))**

We have a topic with the position of the motor. When we remove it (Second screenshot on our previous post) the motor slows down but doesn't stop. We thought that without position feedback the 
`joint_trajectory_controller` would send a stop command to the motor. As I understood we have to send the command to stop ourselves.

**4 - Populating the velocity fields in which read method? Your RobotHW? Did you change the source of the joint_trajectory_controller?**

No, we were talking the RobotHW.

**5 - An older documentation suggests that when filling the velocity fields in the command, the controller would run a quadratic interpolation instead of a quintic one, which may explain the different behaviour, but I'm unsure about how relevant that is to the current controller. You are only expected to fill the position fields, velocity is not needed (have to check the code if the quintic/quadratic switch is still there).**

We will check it. But for now we will only fill out the position.