ros-controls/ros_control #248: [rqt_controller_manager] needs update to Qt5 with Kinetic

wjwwood, 2016-08-17:
When running this rqt plugin in Kinetic I get:

```
RosPluginProvider.load(rqt_controller_manager/ControllerManager) exception raised in __builtin__.__import__(rqt_controller_manager.controller_manager, [ControllerManager]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 77, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_controller_manager/controller_manager.py", line 35, in <module>
    from python_qt_binding.QtGui import QCursor, QFont, QFormLayout, QHeaderView,\
ImportError: cannot import name QFormLayout

PluginManager._load_plugin() could not load plugin "rqt_controller_manager/ControllerManager":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 83, in load
    raise e
ImportError: cannot import name QFormLayout
```

I think it needs a similar update as https://github.com/ros-controls/ros_controllers/pull/231

bmagyar, 2016-08-18:
On it.

bmagyar, 2016-08-18:
https://github.com/ros/rosdistro/pull/12407

Thanks for reporting!

closed 2016-08-18 by bmagyar

wjwwood, 2016-08-24:
Thanks for fixing it :smile:

bmagyar, 2016-08-24:
My pleasure :+1: