ros-controls/ros_control #256: Controller Spawn Issue

sauronalexander, 2016-12-01:
Hello there. I am using ROS Indigo with Gazebo2. I am currently facing a problem that the controller_manager has correctly loaded the controllers but cannot start the controllers. When trying to start the controllers manually, the system gives me "rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details". Can anyone tell me how to resolve this? Thank you!

bmagyar, 2016-12-01:
This seems like a malformed service call to me. Try the quick way with
```
rosrun rqt_controller_manager rqt_controller_manager
```

sauronalexander, 2016-12-01:
I haven't used rqt before. Does the package rqt_controller_manager has an executable? The system seems keeping telling me that there is no executable for rqt_controller_manager.

bmagyar, 2016-12-02:
That's right, it seems we forgot to add the install rule for that in Indigo, I tested last night on Kinetic and there it's being properly installed.

Grab it from here, place in package, chmod it properly, you know the drill ;) 
https://github.com/ros-controls/ros_control/blob/indigo-devel/rqt_controller_manager/scripts/rqt_controller_manager

I'll add a PR to add this install rule

sauronalexander, 2016-12-02:
Thanks. Do I need to install some python packages to run this script? It seems that I am missing QtWidgets here.

sauronalexander, 2016-12-03:
It seemed this problem is resolved through hacking. 

For my case, when compiling source files in workspace, I changed the python_qt_binding.QtWidgets to python_qt_binding.QtGui in src/controller_manager.py. 

Or by installing from Debian, I worked this around by changing the same syntax in rqt_py_common/plugin_container_widget.py.

I do not know the reason behind this since I am not a python guy. I also have no idea whether this hack will work on others'. Any idea why this work will help. Thanks.

sauronalexander, 2016-12-03:
The result when using the rqt_controller_manager is unfortunately, segmentation fault. That's the only output.

bmagyar, 2016-12-03:
Hacking stuff especially in installed packages will only lead to a lot of pain in the future, don't do it. Probably something is missing from your installation.
Aren't you in PAL with someone around you could just grab and get this sorted quick? 

If not, can you just post the command that produced the starting post's error message?

sauronalexander, 2016-12-07:
Hello, I am not at PAL and I am working on my senior thesis at US. So I think the problem is solved simply by reinstalling the ROS. I have been spending the last two days re-configuring everything and it seems it worked with the new system. Strange as it is, I still do not know the reason behind the issue. Thank you for your help anyway.

closed 2016-12-07 by sauronalexander