ros-controls/ros_control #260: TransmissionInterfaceLoader cannot find a handle of JointStateInterface

canalteks, 2016-12-13:
Dear developers,

I found that transmission_interface::TransmissionInterfaceLoader will throw an exception in loading new hardware_interface/Position,Velocity,EffortJointInterfaces from URDF::transmission tags.

This problem only occurs when the users used their own RobotHw with JointStateInterface.

It's because that
- [transmission_interface::VelocityJointInterfaceProvider](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/src/velocity_joint_interface_provider.cpp), and also Position,EffortJointInterfaceProvider, are using [joint_interfaces.joint_state_interface](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/src/velocity_joint_interface_provider.cpp#L68) to get a joint state handle from the resouce.
- but unfortunately, if user's RobotHw has its own JointStateInterface, [the code](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/src/joint_state_interface_provider.cpp#L66) of JointStateInterfaceProvider only populates the RobotHw's JointStateInterface but not TransmissioInterfaceLoader::loader_data_.joint_interfaces.[joint_state_interface](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/include/transmission_interface/transmission_interface_loader.h#L134).
- it keeps TransmissioInterfaceLoader::loader_data_.joint_interfaces.joint_state_interface is empty.

Then, the inconsistency of references between two JointStateInterfaces makes this problem.

I'll suggest following approaches,
1. notify in [the comment](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/include/transmission_interface/transmission_interface_loader.h#L284) of the TransmissionInterfaceLoader --- RobotHw should not have its own JointStateInterface to use TransmissionInterfaceLoader,
2. or add [this code](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/src/joint_state_interface_provider.cpp#L49), instead of using [joint_interfaces.joint_state_interface](https://github.com/ros-controls/ros_control/blob/kinetic-devel/transmission_interface/src/velocity_joint_interface_provider.cpp#L68).

```
// test code to make this problem
RobotHw  robot_hw;
RobotTransmissions  robot_transmissions;
JointStateInterface  joint_state_interface;
robot_hw.registerInterface( &joint_state_interface );

TransmissionInterfaceLoader  loader( &robot_hw, &robot_transmissions );

loader.load( urdf );
```

yours sincerely.

littlejohnyang, 2017-05-08:
I am having similar issue here. I have my own RobotHW with JointStateInterface. It'll throw an error "Failed to load transmission 'xxxx'. It contains no valid hardware interface. I can control the motors fine but I will not be able to receive any feedback. The moment I register a JointStateInterface with my RobotHW class, it'll throw this error. But I need it in order to control the group of motors through joint trajectory controller.

ipa-mdl, 2017-05-08:
It took me a while to understand the problem, but I can confim @canalteks's code analysis.
Another problem exists for the other interface types, if you try to register them after running `loader.load`.

As a workaround, instead of registering the interfaces yourself, you could use the interfaces from  `loader.getData()->joint_interface`

jlack1987, 2018-10-18:
I too have found my way to this bug. Gotta give @canalteks a hand in actually tracking down the problem. I just started getting very strange plugin load errors that gave me no indication of what the actual problem was.

ompugao, 2019-09-05:
any updates for this issue? I got a same error in melodic.
How should I avoid this error?

jlack1987, 2019-09-19:
@ompugao the way I get around it is I make load all transmissions containing my custom hardware interfaces before all others. Bit hacky, but it works for me

bmagyar, 2019-10-15:
I've been thinking about this recently working on similar stuff and I am opting for point 2.
What do you guys think of such a change?