ros-controls/ros_control #269: Lunar release

mikaelarguedas, 2017-04-11:
Hi `ros_control` maintainers!

As you may know the next ROS release Lunar Loggerhead is around the corner :tada: 

All dependencies of `ros_control` are now released into Lunar! Please consider making a Lunar release soon to unlock [downstream packages](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22ros_control%22&s=6&r=1).

Thanks!

ipa-mdl, 2017-04-11:
I have restarted the [lunar travis job](https://travis-ci.org/ros-controls/ros_control/jobs/213375251)

ipa-mdl, 2017-04-11:
@mikaelarguedas: `ros-lunar-urdf` is not yet in `ros-shadow-fixed` (but in testing). How often do they get synced?

bmagyar, 2017-04-11:
Thanks for the note!  I've started releasing bits,  all doc &  source jobs
are in already.

As Mathias said,  urdf was missing so far.

mikaelarguedas, 2017-04-11:
@ipa-mdl @bmagyar Indeed the Debian jobs failed yesterday, they are turning over now and should be available for all platforms in shadow-fixed very soon. Sorry about that

ipa-mdl, 2017-04-11:
Tests have passed :tada:

bmagyar, 2017-04-11:
Thanks! I'll double check with a pre-release and start pulling triggers.

mikaelarguedas, 2017-04-11:
Thanks all for the quick release! I'm going to close this then. All jobs are green :tada: (the ARM ones haven't turned over though)

closed 2017-04-11 by mikaelarguedas