ros-controls/ros_control #271: Functions with return type not returning anything

jlack1987, 2017-05-02:
Working on the current release of `kinetic-devel` it looks like there are some compiler warnings generated for a few methods in `InterfaceManager` that have a return type but don't have a return statement. The offenders are [here](https://github.com/ros-controls/ros_control/blob/kinetic-devel/hardware_interface/include/hardware_interface/internal/interface_manager.h#L66), [here](https://github.com/ros-controls/ros_control/blob/kinetic-devel/hardware_interface/include/hardware_interface/internal/interface_manager.h#L77), and [here](https://github.com/ros-controls/ros_control/blob/kinetic-devel/hardware_interface/include/hardware_interface/internal/interface_manager.h#L93). 

Haven't looked into what these methods actually do, so it might technically not be a problem, but since my group develops with `-Werror` on by default it's a pain to have to now go manually ignore it with `-Wno-return-type`. Also seems like kinda bad c++ foo. Thanks!

bmagyar, 2017-05-03:
Hi Jordan,
Long time, no see ;) 

That bit of metaprogramming arrived with `combined_robot_hw` and I am not fully familiar with it. 
Maybe @toliver could help us out on this one?

davetcoleman, 2017-06-20:
+1 to fixing these warnings, i am experiencing them too on Lunar

ipa-mdl, 2017-10-21:
should be fixed in #285 ;)

closed 2018-02-01 by bmagyar