ros-controls/ros_control #277: Add maximum deceleration

alex-gee, 2017-07-06:
So far e.g.
http://docs.ros.org/jade/api/moveit_msgs/html/msg/JointLimits.html
states that min_acceleration = -max_acceleration, as well as ros_control does.

I would like to hear your opinion on something like:
https://github.com/alex-gee/ros_control/commit/e0986eb9a795b9bf0d220128e429cb136611f6f5#diff-6e16e8c4e58a41d2c3b5acfcc6191ee6

This would allow to specify a max_deceleration (aka min_acceleration) independent from max_deceleration. It would still default to the current case so it should be backward compatible.

Any comments?

AndyZe, 2017-11-06:
I would love this.

bmagyar, 2017-11-06:
> alex-gee/ros_control@e0986eb#diff-6e16e8c4e58a41d2c3b5acfcc6191ee6

It'd be most welcome. Tests are needed of course :)

gavanderhoorn, 2019-02-03:
@alex-gee: did you ever get around to submitting a PR for this?

There appear to be some implementations of this "in the wild" (such as [this one](https://github.com/easymov/ackermann_controller/blob/fce99623c80c49bb72204dd75b8966f591fb103e/include/ackermann_controller/speed_limiter.h#L37-L38) and [this one](https://github.com/PilzDE/pilz_industrial_motion/blob/a0d7eb299962e6ff75717d1f34fb0c926c3db6f7/pilz_extensions/include/pilz_extensions/joint_limits_extension.h#L31-L32)), so there seems to be interest / utility.