ros-controls/ros_control #278: Unable to spwan the controllers - Gazebo, ROS Indigo

LauFe, 2017-07-10:
Hello,
I am having a similar issue to one submitted here before, but I am not really able to solve it, everything seems fine to me.
I built my own package with the description model, configuration file, launch file. I can spawn the model and get the main parameters but when I launch the launch file (attached here) an error occurs and it is not able to spawn the controllers. My package is run dependent on ros_control and gazebo_ros_pkgs, but the problem seems to occur with the controller manager.
I have attached the screen of the error, configuration file and launch file.
[image]
[image]
[image]

Thank you for any advice!

bmagyar, 2017-07-10:
I'd also need to see your full URDF as it appears on the /robot_description
parameter.

My first bet is that you don't have the gazebo plugin defined in the URDF.
Unless that is specified there won't be a controller_manager present within
Gazebo managing the joints of your robot.

LauFe, 2017-07-10:
Thank you for your response, I have included the plugin on the top. I thought that just that was needed.
[image]

bmagyar, 2017-07-10:
Could you just post the full URDF as text please?

My next guess is: missing transmissions.

LauFe, 2017-07-10:
Also that should be fine..
[simple_model.txt](https://github.com/ros-controls/ros_control/files/1136058/simple_model.txt)

LauFe, 2017-07-11:
Hi, I kind of sort out the problem for the position controllers of type effort_controllers/JointPositionControllers, however it is not working for the type position_controllers/JointPositionControllers. Am I missing something about that?
Thank you

bmagyar, 2017-07-11:
What were the changes? 
`position_controllers/JointPositionController` will only work with a joint interface of `position`.
The naming scheme we use for controllers is such that for `X_controllers/JointYController` the interface of the controlled joint is `X` while the commands it accepts and controls is `Y`. Any conversion between `X` and `Y` is done through a PID controller.

Multiple joint interfaces are supported but only after Indigo. You can read a lot on that on the links below:
https://github.com/ros-controls/ros_control/issues/75
https://github.com/ros-controls/ros_control/pull/204

LauFe, 2017-07-11:
I removed, recloned and rebuilt everything again, probably my control packages were corrupted and I was using position_controllers/JointPositionController in a wrong way.  I will read those links then, thank you for the help!

bmagyar, 2017-07-12:
:+1: 
I'm closing the issue now, feel free to reopen if the same problem comes up.

closed 2017-07-12 by bmagyar