ros-controls/ros_control #282: The constructor of ForceTorqueSensorHandle

sunbibei, 2017-09-21:
Hello, when I am using the hardware_interface::ForceTorqueSensorHandle, I confuse the self-define constructor. We should not want to change the data in this class, but the parameter list is asked a non-const pointer. The constructor define as follow, 
```
ForceTorqueSensorHandle(const std::string& name,
                          const std::string& frame_id,
                          double* force,
                          double* torque)
```
I don't known that whether it is a bug or not? Thank you.

ipa-mdl, 2017-09-21:
Good catch!

I don't see any reason for this. It is not bug, but a glitch w.r.t. const correctness.

sunbibei, 2017-09-22:
Thanks your response, forgiving my poor English. there is a similar problem to the structure of ImuSensorHandle::Data. Thank you

ipa-mdl, 2017-09-22:
I have added a const ImuSensorHandle constructor as well.
We should not touch `ImuSensorHandle::Data` as it might be used elsewhere.

sunbibei, 2017-09-22:
OK，i see. Thank you

ipa-mdl, 2017-09-22:
Let's keep this issue open until the PR was merged.

PR #283 (2017-09-26): fix constness in ForceTorqueSensorHandle
fixes #282 

I kept the non-const constructor for the sake of ABI compatibility, not sure if it is really needed.
closed 2017-09-26 by bmagyar