ros-controls/ros_control #286: cannot build ros_control in indigo_devel

shantengfei, 2017-10-27:
I git clone the package from here and put it into my catkin_ws/src and catkin_make in catkin_ws but failed. Here is the output:
`/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp: In member function ‘bool ros_control_ur::UrHardwareInterface::canSwitch(const std::list<hardware_interface::ControllerInfo>&, const std::list<hardware_interface::ControllerInfo>&) const’:
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:180:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface
                      ^
In file included from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/resource_manager.h:40:0,
                 from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/hardware_resource_manager.h:36,
                 from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/joint_state_interface.h:33,
                 from /home/shantengfei/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ur_hardware_interface.h:61,
                 from /home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:58:
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:186:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
       controller_it->hardware_interface.c_str());
                      ^
/opt/ros/indigo/include/ros/console.h:343:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:376:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:558:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:214:24: note: in expansion of macro ‘ROS_LOG’
 #define ROS_ERROR(...) ROS_LOG(::ros::console::levels::Error, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                        ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:183:5: note: in expansion of macro ‘ROS_ERROR’
     ROS_ERROR(
     ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:195:30: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
      if (stop_controller_it->hardware_interface
                              ^
In file included from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/resource_manager.h:40:0,
                 from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/hardware_resource_manager.h:36,
                 from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/joint_state_interface.h:33,
                 from /home/shantengfei/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ur_hardware_interface.h:61,
                 from /home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:58:
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:205:23: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
        controller_it->hardware_interface.c_str());
                       ^
/opt/ros/indigo/include/ros/console.h:343:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:376:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:558:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:214:24: note: in expansion of macro ‘ROS_LOG’
 #define ROS_ERROR(...) ROS_LOG(::ros::console::levels::Error, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                        ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:202:6: note: in expansion of macro ‘ROS_ERROR’
      ROS_ERROR(
      ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:209:29: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   } else if (controller_it->hardware_interface
                             ^
In file included from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/resource_manager.h:40:0,
                 from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/hardware_resource_manager.h:36,
                 from /home/shantengfei/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/joint_state_interface.h:33,
                 from /home/shantengfei/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ur_hardware_interface.h:61,
                 from /home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:58:
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:215:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
       controller_it->hardware_interface.c_str());
                      ^
/opt/ros/indigo/include/ros/console.h:343:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^
/opt/ros/indigo/include/ros/console.h:376:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
[... 35 lines cut ...]
      ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp: In member function ‘virtual void ros_control_ur::UrHardwareInterface::doSwitch(const std::list<hardware_interface::ControllerInfo>&, const std::list<hardware_interface::ControllerInfo>&)’:
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:251:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface
                      ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:256:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface
                      ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:267:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface
                      ^
/home/shantengfei/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:272:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface
                      ^
[ 97%] Building CXX object ros_control/controller_manager_tests/CMakeFiles/controller_manager_tests.dir/src/pos_eff_opt_controller.cpp.o
[... cut ...]
make[1]: *** [ur_modern_driver/CMakeFiles/ur_hardware_interface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Building CXX object ros_control/controller_manager_tests/CMakeFiles/controller_manager_tests.dir/src/my_dummy_controller.cpp.o
Linking CXX shared library /home/shantengfei/catkin_ws/devel/lib/libcontroller_manager_tests.so
[ 97%] Built target controller_manager_tests
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
I tried to download another package from the github and catkin_make worked. So is this problem from the package?

bmagyar, 2017-10-27:
Certainly not. Our continuous integration is fine. 

My first guess is that you are trying to build the kinetic-devel branch on Indigo and missing dependencies. 

Why are you building `ros_control` from source in the first place?

shantengfei, 2017-10-28:
I tried to install the apckage by using apt-get,but  the output was **E: Unable to locate package ros-indigo-ros-control E: Unable to locate packa**. Then I tried to download the package into my workspace.Now i have installed using the second method in the wiki.

bmagyar, 2017-10-28:
You have to review your apt setup or the same issue will surface with other packages.
You can find all binaries that apt-get would fetch if it was set up right here, for example: http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-ros-control/

If something that is up there but you can't find/download you are missing the right apt setup or have trouble with the network, accessing things through a proxy for instance.

closed 2017-10-28 by bmagyar