ros-controls/ros_control #297: ros2 release

jlack1987, 2018-01-05:
Am wondering if there's a date in mind to release a version of ros_control that is an ament package for ros2?

Karsten1987, 2018-01-08:
I guess this question targets mainly the maintainer of the existing packages. But I believe you've seen the ros2_control and ros2_controller packages which fully feature ament and can serve as a starting point.

https://github.com/ros-controls/ros2_controllers
https://github.com/ros-controls/ros2_control

jlack1987, 2018-01-08:
oh hey thanks! No I had not seen those

closed 2018-01-08 by jlack1987

bmagyar, 2018-01-11:
Keep in mind, that our current `ros2_control` implementation is a proof of concept. We don't yet have plans for migrating the existing `ros_control` packages to ROS2 but we'd welcome any contribution on that side!