ros-controls/ros_control #298: Switch MultiInterfaceController to be variadic template

mikepurvis, 2018-01-11:
This is possible in Kinetic as C++11 is now the target platform:

http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021

Would clean up the T1/T2/T3/T4 stuff in this file:

https://github.com/ros-controls/ros_control/blob/kinetic-devel/controller_interface/include/controller_interface/multi_interface_controller.h

ipa-mdl, 2018-01-11:
> This is possible in Kinetic as C++11 is now the target platform:

I don't understand it this way. AFAIK everything must be compilable with C++11, but it is not the default target.

mikepurvis, 2018-01-11:
I think that was the status for Jade, which said that the target platform was:

> C++03 (C++11 features are not used, but code should compile when -std=c++11 is used)

Versus Kinetic:

> C++11 (GCC 4.9 on Linux, as it's the version that Debian Jessie ships with)

I think the real issue is that (I guess for compatibility reasons?) even the GCC 5 that Xenial comes with doesn't have C++11 as the default— the features are still behind a flag.

ipa-mdl, 2018-01-11:
> C++11 (GCC 4.9 on Linux, as it's the version that Debian Jessie ships with)

This means that if you'd like to enforce c++11 support you can now do so for platforms, e.g. with a flag.

In addition there is the [rationale on c++](http://www.ros.org/reps/rep-0003.html#c) with recent fix https://github.com/ros-infrastructure/rep/pull/152 :

> Use of C++11/C++14 features and filesystem/networking/etc... TS's (Technical Specifications) is allowed if they are checked for at configure time and equivalent functionality can be provided without the extra compiler features. 

This will get changed for melodic (https://github.com/ros-infrastructure/rep/pull/139)