ros-controls/ros_control #300: ControllerManager could use a better shutdown sequence

mikepurvis, 2018-01-17:
Issue is when using `spawner` in a launch file, it often happens that the base node is able to exit prior to the spawner having a chance to unload the controllers, so it always has to time out and spew this warning:

```
[robot-1] killing on exit
[INFO] [/base_controller_spawner] [1516211781.095033]: Shutting down spawner. Stopping and unloading controllers...
[WARN] [/base_controller_spawner] [1516211783.307672]: Controller Spawner couldn't reach controller_manager to take down controllers. Waited for X second(s).
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Where the default for X is like 30 seconds, so roslaunch ends up SIGTERMing it unless you drop the default.

One way to improve this might be to make `ControllerManager::getControllerNames` public, so that the control loop can avoid exiting until all controllers have been stopped/unloaded, or provide a new `ControllerManager::getControllerCount` method, eg:

```
void controlLoop()
{
  while(ros::ok() || cm_.getControllerCount() > 0)
  {
    robot_.read();
    cm_.update(t, d);
    robot_.write();
  }
}
```

This ensures that the controllers have had a chance to perform a controlled shutdown before the loop is stopped.

ipa-mdl, 2018-01-17:
I am okay wth exposing these interfaces.

However, I would not recommend to use `spawner` as part of the bringup.
IMHO `spawner` and `unspawner` are meant to be used in separate launch files.
For bringup `controller_manager spawn` works fine.

Your proposed loop works as long as only `spawner` was used to start controllers.
But as soon as someone uses rqt or calls the service directly (=no automatic clean-up), then the 
`controller_manager` node gets the 30s penalty.

mikepurvis, 2018-01-17:
Ah, I misunderstood the intended usage— that makes sense.

Will close this for now.

closed 2018-01-17 by mikepurvis