ros-controls/ros_control #304: Spawer is unable to shutdown properly after a controller has been changed

harry17189, 2018-03-07:
With our setup, we start ros_control in a joint_trajectory_controller to move it into a standard starting configuration.
Afterwards we use a rospy service call to switch to a custom controller.

when stopping the ROS system, the spawner then wants to stop all controllers it knows that are running.
However, it doesn't actually look up which controllers are running at the moment.

I'm unfamiliar with the details, but is it possible for the shutdown function of the spawner to get the current list of running controllers and stop en unload these instead of blindly trying to stop and unload the controllers it started

For example:
file :  ros_control/controller_manager/scripts/spawner
```python
def shutdown():
    global loaded,unload_controller_service,load_controller_service,switch_controller_service

    rospy.loginfo("Shutting down spawner. Stopping and unloading controllers...")

    try:
        # unloader
        unload_controller = rospy.ServiceProxy(unload_controller_service, UnloadController)

        # switcher
        switch_controller = rospy.ServiceProxy(switch_controller_service, SwitchController)

        rospy.loginfo("Stopping all controllers...");

        # insert reading of running controllers here
        loaded = funtionToReadRunningControllers

        # Switch off all running controllers
        switch_controller([], loaded, SwitchControllerRequest.STRICT)
        rospy.loginfo("Unloading all loaded controllers...");
        for name in reversed(loaded):
            rospy.logout("Trying to unload %s" % name)
            unload_controller(name)
            rospy.logout("Succeeded in unloading %s" % name)
    except (rospy.ServiceException, rospy.exceptions.ROSException) as exc:
        rospy.logwarn("Controller Spawner error while taking down controllers: %s"  % (exc))
```

Is this possible ?

ipa-mdl, 2018-03-07:
The `spawner` script is meant to take care of the controller over the full lifecycle.
It looks like you want to use `controller_manager spawn ...` instead.

harry17189, 2018-03-07:
I think I see my problem.

In my launch file I'm launching two spawners.
One for initial running controllers and a second for the stopped controllers
This to prevent a resource conflict.
```xml
<node name="controller_spawner" pkg="controller_manager" type="spawner" args="joint_state_controller arm_state_controller joint_trajectory_controller --shutdown-timeout 2"/>
<node name="controller_stopper" pkg="controller_manager" type="spawner" args="--stopped custom_controller --shutdown-timeout 2" />
```
In our script we switch from the joint_trajectory_controller to the custom_controller.

From the [wiki](http://wiki.ros.org/controller_manager?distro=kinetic#1.2)
> 1.2 spawner
To automatically load and start a set of controllers at once, and automatically stop and unload those same controllers at once, use the spawner tool:
>    `$ rosrun controller_manager spawner [--stopped] name1 name2 name3`
> When you run spawner, the listed controllers will get loaded and started (unless you specify --stopped). Spawner will keep running while the controllers are up. When you kill spawner (ctrl-c) it will automatically stop and unload all controllers it initially started.

As a result when killing the system (Ctrl-c)  both spawners will try and stop their ***initially started*** running controllers.
However the first spawner can't stop a controller that isn't running anymore and throws an error.

This could be fixed by using one spawner, but is it possible to use running and stopped controllers in the arguments?

@ipa-mdl I don't know what you mean with the call to `spawn`. This is an executable ? is there documentation? I look for the source code but looking at the Cmake, this isn't built. Please elaborate

ipa-mdl, 2018-03-07:
You could run it like this:
```xml
<node name="controller_spawner" pkg="controller_manager" type="controller_manager" args="spawn joint_state_controller arm_state_controller joint_trajectory_controller"/>
<node name="controller_loader" pkg="controller_manager" type="controller_manager" args="load custom_controller"/>
```

Both will wait for the services, spawn/load and exit afterwards (without unloading and stopping).