ros-controls/ros_control #305: Changes for Melodic

bmagyar, 2018-03-20:
Hello @ros-controls/ros_control team,

I'll be updating this issue with TODO-s for the Melodic release.
We are expecting some changes to break ABI and probably API even after the first release of Melodic. Some project have unlucky timing.

* [ ]  class_loader headers migration
* [ ]  pluginlib headers migration

These changes should affect `ros_control` and `ros_controllers`.

ipa-mdl, 2018-03-20:
ABI changes do not matter, it is quite hard to mix kinetic binaries with melodic binaries and the ABI won't be compatible anyway (C++14..).
There are no API breaks planned for melodic (?).
To get rid of the warnings, just the new header names should be used.
The new headers are already available for indigo and kinetic (shadow-fixed)..

ipa-mdl, 2018-03-20:
`ros_control` (ittself) can be built and run with melodic using https://github.com/ipa-mdl/ros_control/commit/d63e833f290bda15ca0352cf1248a9eae30bdb0e
Travis: https://travis-ci.org/ipa-mdl/ros_control/jobs/356053823

ipa-mdl, 2018-03-22:
@bmagyar: I would suggest to release `ros_control` rather early, there are no blockers anymore.

bmagyar, 2018-03-22:
Thanks for the heads up, it moves faster than I expected!

bmagyar, 2018-03-30:
Initial release was made. I've just realised we are still using `boost::shared_ptr`. That should change to `std::shared_ptr` since starting from Melodic C++11 is default.
Any thoughts @ros-controls/ros_control ?

ipa-mdl, 2018-03-30:
>  since starting from Melodic C++11 is default.

C++14 (default of g++6.1 and above).

> That should change to std::shared_ptr

I would not change it directly.
This would force all dependent packages to branch for melodic.
Instead all pointers should get types in kinetic and melodic.
These can be switched to `std` pointers for new versions afterwards.

ipa-mdl, 2018-03-31:
@bmagyar: What convention do you prefer: `MyType[Const]Ptr` or `MyTypeShared[Const]Ptr`?
The latter is longer, but more specific wrt the shared ownership.

bmagyar, 2018-03-31:
Personally the latter. The question is if we use already a few that follow
the shorter version. Urdf maybe? Or should we care at all?

ipa-mdl, 2018-03-31:
After a brief review:
URDF uses `MyType[Const]SharedPtr`.
In the ros-controls repositories `MyType[Const]Ptr` is used.

bmagyar, 2018-04-13:
Agreed!

bmagyar, 2018-04-17:
#324