ros-controls/ros_control #316: Return policy mismatch between getJointLimits() and getSoftJointLimits()

yoshito-okada, 2018-04-09:
I am wondering why joint_limits_interface::getSoftJointLimits() returns false when no related parameters set although getJointLimits() returns true in the same situation.

I think returning true would be natural because these functions are intended to update the given limits as much as possible on the basis of parameters. So I would propose to remove [L225](https://github.com/ros-controls/ros_control/blob/16499ff46d3294ea32c0583be9da2b5c3bde5eb8/joint_limits_interface/include/joint_limits_interface/joint_limits_rosparam.h#L225) and to edit [L229](https://github.com/ros-controls/ros_control/blob/16499ff46d3294ea32c0583be9da2b5c3bde5eb8/joint_limits_interface/include/joint_limits_interface/joint_limits_rosparam.h#L229) to "return true" in joint_limits_rosparam.h.

miguelprada, 2018-04-11:
I see your point. However, I still think it would be nice to return false in case `has_soft_limits` is set and a complete soft limit specification can not be found. What do you think about:

```c++
inline bool getSoftJointLimits(const std::string& joint_name, const ros::NodeHandle& nh, SoftJointLimits& soft_limits)
{
//...
  // Override soft limits if complete specification is found
  bool has_soft_limits;
  if(limits_nh.getParam("has_soft_limits", has_soft_limits) && has_soft_limits)
  {
    if(!limits_nh.hasParam("k_position") || !limits_nh.hasParam("k_velocity") || !limits_nh.hasParam("soft_lower_limit") || !limits_nh.hasParam("soft_upper_limit"))
    {
      return false;
    }
    limits_nh.getParam("k_position", soft_limits.k_position);
    limits_nh.getParam("k_velocity", soft_limits.k_velocity);
    limits_nh.getParam("soft_lower_limit", soft_limits.min_position);
    limits_nh.getParam("soft_upper_limit", soft_limits.max_position);
  }

  return true;
}
// ...
```

It could be argued that nothing like this happens when parsing joint limits if, for instance `has_position_limits == true` and `min_position` or `max_position` are not set. Consistency with this behavior favors your proposed solution, @yoshito-okada. I still find it weird to return true if such an inconsistency in the parameters is found.

Thoughts, @ipa-mdl ?

yoshito-okada, 2018-04-11:
Almost agree.

I think users are likely to expect `get*JointLimits()` functions to return whether it successfully copied parameters to instances because their names begin with `get` (I know current implementation is not so but well documented).  From this viewpoint, when no `has_soft_limits` on the parameter server and the instance not edited, the function should return true as @miguelprada 's suggestion.

However, when `has_soft_limits: false` on the server and the instance not edited, is it successful reflection from parameters to the instance?  Perhaps an additional solution is;
- add `has_soft_limits` variable to `SoftJointLimits` 
- set `has_soft_limits = false` and `return true` in `getSoftJointLimits()` when `has_soft_limits: false` on the server

miguelprada, 2018-04-12:
I think the reasonable thing to do is what @yoshito-okada suggested in his original post, since it's the most consistent behaviour with `getJointLimits`.

Whether it is more reasonable to return false if incoherent parameter specifications are found in either the soft or regular limit parameters (e.g. `has_position_limits: true` but no `min_position`) is probably a different matter and can be discussed in a different issue for regular and soft limits together.

Should a PR about this target `kinetic-devel` or `melodic-devel`?

yoshito-okada, 2018-04-12:
I think returning false on incoherent parameter is reasonable.  But `getSoftJointLimits()` still has an issue which there is no way to notify soft limits are deactivated (e.g. `has_soft_limits: false` on the param server) because either returning true or false is already used to represent other cases.

On the other hand, `getJointLimits()` does not have this issue because it does `JointLimits::has_*_limits = false` .  This is why I proposed to add `SoftJointLimits::has_soft_limits` in my previous post with consideration of consistency.

As @miguelprada pointed out, this may be a different issue.  I never mind to close this.

miguelprada, 2018-04-13:
> I think returning false on incoherent parameter is reasonable. But getSoftJointLimits() still has an issue which there is no way to notify soft limits are deactivated (e.g. has_soft_limits: false on the param server) because either returning true or false is already used to represent other cases.

That's a very good point. The only issue I see in adding `SoftJointLimits::has_soft_limits` is that no `RobotHW` implementation would currently support it and it may end up being a source of confusion.

I do think that this discussion should belong to this issue. Even if it is not directly related to its return value (as stated in the issue name), it is related to making the behavior of `getSoftJointLimits` as similar as possible to `getJointLimits`.

yoshito-okada, 2018-04-13:
I understand adding `SoftJointLimits::has_soft_limits` would require changes not only in `joint_limits_interface` but also `hardware_interface` and maybe `urdf` .  However I still think the issue that current `SoftJointLimits` cannot represent "no limits" state although urdf and rosparam can (by no `<safety_controller>` tags and `has_soft_limits: false`) should be solved.

yoshito-okada, 2018-04-13:
Understood.