ros-controls/ros_control #317: Joint role tag is not parsed from urdf but it exists in the JointInfo struct

jlack1987, 2018-04-11:
The `transmission_interface::JointInfo` struct has `role_` as a data member but the transmission parser doesn't parse it.

traversaro, 2018-04-17:
A naive question: is the semantics of this `<role>` tag documented somewhere? It is supposed to be documented in http://wiki.ros.org/ros_control#Transmission_URDF_Format or http://wiki.ros.org/urdf/XML/Transmission ?

jlack1987, 2018-04-17:
I don't believe it is documented to my knowledge. Only way I knew it was there was just time spent digging through the code.

bmagyar, 2018-04-20:
TBH I am not aware the original intention of it so I think it is up for us to define it now.
I'd add documentation to *both* places. The one at URDF is not disconnected from us, was written by one of our maintainers.

jlack1987, 2018-04-20:
Well just as a use case example for the waist/ankle transmissions on the Valkyrie robot we used the `role` tags to tell the transmissions which actuator is which(left/right) because the solution to the transmission needs to know which actuator the positions/velocities/efforts correspond to

PR #318 (2018-04-17): Update transmission parser to parse the joint role
This is to fix #317