ros-controls/ros_control #320: realtime_tools are dependencies but not actually used

kjaget, 2018-04-13:
This is in the kinetic release tag release/kinetic/controller_manager/0.13.1-0 (and probably earlier as well).  The dependencies show up in CMakeLists.txt, package.xml and controller_manager.h in controller_manager and then also in ros_control/package.xml.

bmagyar, 2018-04-13:
Nice catch, `controller_manager` probably doesn't need to depend on `realtime_tools`. The dependency in `ros_control` is there since that's a metapackage, `realtime_tools` is part of the `ros_control` bundle, every controller uses it for instance.

kjaget, 2018-04-13:
Ah, makes sense. I missed the obvious point about ros_control being a meta-package.

bmagyar, 2018-04-27:
Resolved in the referenced PR. Thanks for the report!

closed 2018-04-27 by bmagyar