ros-controls/ros_control #333: #define in system header netdb.h conflicts with transmission loader enum

jlack1987, 2018-05-06:
For some context, i'm trying to use boost asio in conjunction with the ros control RobotHw and there are of course transmissions involved. In trying to use both `transmission_interface` and `boost::asio`(via the `#include <boost/asio.hpp>` header), I have discovered a conflict between a `#define` in `netdb.h` which is used by boost and an enum defined in `transmission_loader.h`. Long story short if you have Boost installed there'll be a dependence on `netdb.h` and in that file is the following line,
```
# define NO_DATA	4	/* Valid name, no data record of requested`
```
and in `transmission_loader.h` there's this enum defined,
```
enum ParseStatus
  {
    SUCCESS,
    NO_DATA,
    BAD_TYPE
  };
```
which is a collision and you can't compile.

bmagyar, 2018-05-07:
And you need this fixed on Kinetic? (It'd be easier to fix on melodic with
C++11.)

jlack1987, 2018-05-07:
Yeah we are still using kinetic, will prob look to switch when the 18.04 stuff comes out. Out of curiosity how do you fix it without changing the name of the `NO_DATA` enum and possibly breaking compatability where ppl are using this? I have hacked around it by doing the following,
```
#undef NO_DATA
#include <transmission_interface/transmission_loader.h>
#define NO_DATA 4
```

bmagyar, 2018-05-07:
I would use a class enum

http://en.cppreference.com/w/cpp/language/enum

Yes it is correct that it probably breaks downstream code but for Melodic
that is fine. That's what the changelog is for.

ipa-mdl, 2018-05-15:
> This would still lead to the naming conflict (?) as the the macro would get expanded in the enum declaration. However, this might the depend on the order.

http://cpp.sh/4275

For unscoped enums it depends on the order.  The `undef` trick possibly breaks the ABI (enum value gets changed to another value in certain source files), same for reordering.
With scoped enums the code will never compile.

closed 2020-08-28 by bmagyar