ros-controls/ros_control #338: How to get control outputs?

sahandrez, 2018-05-30:
Hi, 
I am pretty new to ROS_control, so this might be an easy question. 
I am working on a robot which is controlled with ros_control package. How can I get the control outputs (the efforts sent to the joints)? I looked in to the topics, and wasn't able to find any related topic.
Thanks in advance.

bmagyar, 2018-05-31:
Hi, 

Depending on which controller you are using you may or may not have access to this directly. Some controller publish their commands separately for debugging or something close to that (e.g. the `diff_drive_controller` can publish the modified twist). 

If you really need these values I would publish these directly from the RobotHW of the platform as that is the one place all controller values are collected & sent out to hardware. Typically you can find this in the same node which advertises the `/controller_manager/*` services. If you don't have access to this code, you may try to modify the specific controller you are using and publish the control outputs in the `update()` function of said controller. Make sure to use a publisher from `realtime_tools` for this to keep the controller realtime-safe.

closed 2018-05-31 by bmagyar

sahandrez, 2018-05-31:
Thank you very much for your answer.

By the way, I have had posted this question on ROS answers about a week ago, [here](https://answers.ros.org/question/292178/how-to-get-control-outputs-from-ros_control/)'s the link, but unfortunately no body answered. Thanks again.