ros-controls/ros_control #341: controller_manager does not print warnings when strictness is set to default value

jgueldenstein, 2018-07-03:
When spawning a controller without specifying the strictness, it defaults to best effort with the following warning: 
``` Controller Manager: To switch controllers you need to specify a strictness level of controller_manager_msgs::SwitchController::STRICT (2) or ::BEST_EFFORT (1). Defaulting to ::BEST_EFFORT. ```

When an error occurs when starting a controller this way, no warning is printed.
When loading a controller with strictness "STRICT" an error like 
```Could not start controller with name foo because no controller with this name exists```
is printed.
 
A warning should be printed when an error occurs, even with `BEST_EFFORT`.
This would help debugging even simple typos when the strictness was not explicitly set or "BEST_EFFORT" behavior is requested but information about errors during controller loading is still desirable.

We could provide a patch for this but would like to ask for your opinion before.

@v4hn

bmagyar, 2018-07-04:
Sounds like a good catch, please go ahead with a PR when ready ;)

ipa-mdl, 2018-07-04:
According to the [code](https://github.com/ros-controls/ros_control/blob/6d8b0d27240e908c244cc62d99b95c8847b46b00/controller_manager/src/controller_manager.cpp#L401) a debug message will get logged, but only for this particular case.

I am okay with changing this to warning and adding the output for the other branches.
(Bonus points for using `ROS_LOG_STREAM`)