ros-controls/ros_control #346: View Available Hardware_Interface Handles

TomShkurti, 2018-07-10:
It would be helpful both for debugging and for familiarization with new robots (in cases where documentation is dubious or lacking) if there was a way to list all exposed hardware_interface handles and their types, similar to the controller_manager's `list` and `list-types` options.

bmagyar, 2018-07-11:
That is indeed something I had in mind for a long time now.
We'd need to add support for this to `RobotHW` and `CombinedRobotHW`.

littlejohnyang, 2018-07-17:
Yes I think this is definitely useful. I'm currently running into a problem that my own controller is complaining missing hardware interface (which I created myself). I really wanna see if it's actually available or not in order to debug what I have done wrong.

bmagyar, 2018-08-16:
You can find this in one of the resourcemanager instances within the controller_manager, it should be easy to expose.