ros-controls/ros_control #347: Controller and Hardware Interface for I/O application

piyush2307, 2018-07-18:
All industrial robots uses I/O to communicate with external environment. If I/O's are provided in controller level, it will be easy to control trajectory with I/O. I/Os can be digital or analog.

ipa-mdl, 2018-07-18:
I have started to implement a prototype (msgs, HW interface, controller), but I had to postpone it for now.

piyush2307, 2018-07-21:
can you link it here

gavanderhoorn, 2019-01-12:
@ipa-mdl @mikepurvis: did you guys end-up taking this any further? I'm in a project that would need something similar and I'd like to avoid duplicating your efforts.

My underlying hw is not register-based, but that is an implementation detail.

mikepurvis, 2019-01-13:
@gavanderhoorn We've done some work on it, but it's not in a publicly-releasable state. If you're interested to take a look and see if it's a design direction which interests you, I can upload it to a private github repo and you can take a look.

gavanderhoorn, 2019-02-03:
@mikepurvis wrote:
> If you're interested to take a look and see if it's a design direction which interests you, I can upload it to a private github repo and you can take a look.

Late reply, but yes, I would be interested.

farshbafdoustar, 2019-11-25:
Hi @gavanderhoorn, @ipa-mdl and @mikepurvis: 
Is there any news about these I/O HW interfaces?

I have already started wraping Profinet I/O and Profinet Drive(Telegram 111 and Telegram 3) profiles as ROS_CONTROL HW interfaces in a project to make a general Profinet HW interface compatible to existing position & velocity interfaces + non existing I/O interfaces.  

That would be nice to know if there is any further advancement on this topic to be compatible with that and to avoid duplicating your efforts + of course contributing to what you have already started.

gavanderhoorn, 2019-11-25:
No, I don't have any news.

The offer by @mikepurvis was/is interesting, but I don't believe he found the time to upload what he referred to.

I'd still be interested.

>  have already started wraping Profinet I/O and Profinet Drive(Telegram 111 and Telegram 3) profiles as ROS_CONTROL HW interfaces in a project to make a general Profinet HW interface compatible to existing position & velocity interfaces + non existing I/O interfaces.

do you have any repository where you're doing this?

I'd be interested to follow your development.