ros-controls/ros_control #348: Kinetic : Hardware Interface :prepareSwitch, canSwitch, doSwitch not happening

piyush2307, 2018-07-31:
I have two mode to run , position and effort. While working in indigo version I was using canSwitch(), and doSwitch (defining the mode in this). But in kinetic , controller manager is not calling the canSwitch and doSwitch. Can I get example for this ?

ipa-mdl, 2018-07-31:
`canSwitch` has been deprecated and was removed eventually (https://github.com/ros-controls/ros_control/commit/4c97a05d833291e4942193d16cac5722b1e824d7). Please use `prepareSwitch` instead.

piyush2307, 2018-08-01:
How do I switch controller such that hardwareInterface also switches from position interface to velocity interface. I am using 
```
rosservice call /coro/controller_manager/switch_controller "start_controllers:
- 'position_trajectory_controller'
stop_controllers:
- ''
strictness: 0"
```
to switch the controller, it is switching the controller, but not the hardware interface

```
void HardwareInterface::doSwitch(const std::list<hardware_interface::ControllerInfo> &start_list,
				    				     const std::list<hardware_interface::ControllerInfo> &stop_list)
	{	
		ROS_INFO_STREAM_NAMED(name_,"do switch");

		for (std::list<hardware_interface::ControllerInfo>::const_iterator controller_it =
			stop_list.begin(); controller_it != stop_list.end();
			++controller_it) {
			if (controller_it->type
					== "hardware_interface::VelocityJointInterface") {
				velocity_interface_running_ = false;
				ROS_INFO_STREAM_NAMED(name_,"Stopping velocity interface");
			}
			if (controller_it->type
					== "hardware_interface::PositionJointInterface") {
				position_interface_running_ = false;
				ROS_INFO_STREAM_NAMED(name_,"Stopping position interface");
			}
		}
		for (std::list<hardware_interface::ControllerInfo>::const_iterator controller_it =
				start_list.begin(); controller_it != start_list.end();
				++controller_it) {
			if (controller_it->type
					== "hardware_interface::VelocityJointInterface") {
				velocity_interface_running_ = true;
				ROS_INFO_STREAM_NAMED(name_,"Starting velocity interface");
			}
			if (controller_it->type
					== "hardware_interface::PositionJointInterface") {
				position_interface_running_ = true;
				ROS_INFO_STREAM_NAMED(name_,"Starting position interface");
			}
		}
		
	}
```

ipa-mdl, 2018-08-01:
> How do I switch controller such that hardwareInterface also switches from position interface to velocity interface.

`canSwitch`  and ` prepareSwitch` are mostly the same, but the latter is not const anymore (#218).
If you need any support in inplementing the switching for your specific hardware please ask on [ROS answers](https://answers.ros.org).