ros-controls/ros_control #351: Joint state controller sending wrong velocities when using joint effort controller

sahandrez, 2018-08-16:
Hello Everyone, 

I am working with Kinova Jaco 2 arm in Gazebo. I have set up joint effort controllers (type: `effort_controllers/JointEffortControllers`) for each joint as well as a state publisher (type: `joint_state_controller/JointStateController`) to publish the joint states. When Gazebo launches the robot falls down and all of the torques are set to zero (which is correct) but surprisingly the joint velocities are large non-zero values. 

This is an example of the joint states:

    header: 
      seq: 374
      stamp: 
        secs: 7 
        nsecs: 960000000 
      frame_id: '' 
    name: ['j2n6s300_joint_1', 'j2n6s300_joint_2', 'j2n6s300_joint_3', 'j2n6s300_joint_4', 'j2n6s300_joint_5', 'j2n6s300_joint_6', 'j2n6s300_joint_finger_1', 'j2n6s300_joint_finger_2', 'j2n6s300_joint_finger_3']
    position: [-0.9012024910200029, 1.0208960902473638, 2.5683548384732386, -3.113053974961307, -3.003218016745004, -14.645901287670581, 0.49310698169504263, 1.9998755950416065, 1.9992591337701349] 
    velocity: [22.471898998067832, -9.565097861942068, 0.8652258587505691, 1.1186389825126324, -1.176501196002201, -0.4477771946902821, -0.3705619297587054, 0.2368315633849316, 0.5800874302398775] 
    effort: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0] 

An interesting point is that Gazebo will publish correct joint velocities if I switch the controllers to `effort_controllers/JointPositionController`. 

Does anybody know the reason behind this strange behaviour? How can I fix it? While googling I found there were some issues with joint state controller in the past.
[zero joint velocities from gazebo_ros](https://github.com/jacknlliu/development-issues/issues/49)
[reset /joint_states and gazebo plugins state in gazebo ](https://github.com/jacknlliu/development-issues/issues/27)

Thanks a lot, <br />
Sahand

bmagyar, 2018-08-18:
I can only recite the answer I gave to those issues. Take a look at [gazebo_ros_control](https://github.com/ros-simulation/gazebo_ros_pkgs/tree/kinetic-devel/gazebo_ros_control/src) as these values come from there. The controller is only responsible for converting high level commands to joint handle commands. This is later thrown into the dynamics engine of Gazebo and the values Gazebo returns are then outputted to the state handles and later to `/joint_states`. I see there has been some improvement on this front lately there, see if the version you are running has the latest changes (version number) or grab it from sources and see if that works better for you.

Alternatively, perhaps the dynamics properties are not entirely stable for the dynamics engine (zero inertias on a link or ill-defined mass/CoM definition) in the URDF of the Jaco arm.

sahandrez, 2018-08-20:
Thank you very much for your answer. 
Right now, I believe the dynamics properties are not stable for the engine. I may need to change them. From your experience, which properties do you think more essential for a stable simulation?

Thanks,
Sahand

bmagyar, 2018-08-20:
The worst is usually zero inertias on any link.