ros-controls/ros_control #354: The difference between ros_control or rtt_ros_control?

qkx515, 2018-09-09:
My ultimate goal is to implement real-time control in ros in order to make biped balance. And, I am relatively new to real-time control. 

---
In [**ROSCon-2014: ros_control: An overview**](https://vimeo.com/107507546), Adolfo explains the whole idea about ros_control. It is said that you can achieve real-time control using ros_control. 

[image]

From my point of view, I can get IMU data in real-time and receive data and send commands to controller in real-time, then I shall be able to balance biped in real-time. Am I right? 

---

However, I watched another roscon video - [**ROSCon-2017: How to design ROS-powered robots**](https://vimeo.com/236175447). He said that, "ros_control is component that running in orocos"

[image]

What does that suppose to mean?

Do we really need to install orocos in order to use ros_control?

---

Also, there are 2 repos related to ros_control. Could anyone please explain their differences? 

```
https://github.com/orocos/rtt_ros_control
https://github.com/ros-controls
```

Thank you all in advance. :-)
Kaixian QU

bmagyar, 2018-09-12:
In short, ros_control is a software framework leveraging parts of ROS to implement and manage controllers. When implementing ros_control for a robot, one can choose to do it using a realtime environment which is supported with all controllers provided by the ros_controllers repository. 

The actual bread and butter of your realtime hardware interfacing should be part of your RobotHW implementation (also referred to as the Hardware Abstraction Layer) . For every robot this component is typically hosted in robot_name_hardware, (check out  jackal_hardware or gazebo_ros_control). You won't find many of these openly available as it usually directly interacts with the hardware which manufacturers often prefer to keep private.

Orocos is the library PAL robotics is using to implement their RobotHW.

qkx515, 2018-09-12:
Thank you! I understand partially what you are saying.  However, there are still some questions.

1. Are you saying when implementing RobotHW, we need to use both ros_control framework and orocos, or we just need Orocos?

[image]

[image]

2. What is the purpose of rtt_ros_control? Did the rtt_ros_control integrate ros_control and orocos, so that we can wrote our RobotHW and controller more easily? 
[image]

Thank you for your patience.  :-)

bmagyar, 2018-11-08:
Sorry for the late reply.

Think of the dashed-dotted lines not as hard-defined boundaries but overlapping points. With that mindset your two figures take the same shape. 

The answers are:
1. Yes.
2. Yes.

qkx515, 2018-11-10:
Thanks for the help! :-)

closed 2018-11-10 by qkx515