ros-controls/ros_control #356: Kinetic: Error in controller spawn: dynamic_reconfigure

JavierLRH, 2018-09-16:
Hi everyone I have a problem when I spawn the controller.

In one computer the code works without problems with ROS Kinetic 1.12.7, but in another computer Dynamic Reconfigure is called and change all the parameters.

[image]

I have tried to write the code in many ways but the result is the same.

 Both computers are running Ubuntu 16.04 x64, and the code is the same.

Thanks.

artivis, 2018-09-26:
Hi @JavierLRH,
could you provide a little more information about what you expect and what you actually see ??
Moreover there must be a difference between the two computers if you experience an issue on one but not the other. Any idea ??

JavierLRH, 2018-09-29:
Hi @artivis, thansk for the answer.

I already find the "problem". In the computer with the Warning message, I installed a new version of diff_drive_controller who added the dynamic_reconfigure server in the code.

The dynamic_reconfigure, only changes the multiplier of the wheel radius and wheel separation, not the actual radio. The message is a little bit confusing.

[image]

Regards

closed 2018-09-29 by JavierLRH