ros-controls/ros_control #365: ControllerManager::switchController blocks ros::shutdown

canalteks, 2018-10-19:
Dear developers,

I found that ControllerManager will block ros::shutdown under following conditions.

1. invoke a node with ControllerManager and a node of control_manager or spawner in a **single-roslaunch file**.
2. after loading some controllers via control_manager and calling switchControllerSrv service, shutdown the roslaunch file (with Ctrl+C) **before** calling ControllerManager::update().
3. then ros::shutdown was called, but it was **never** finished, because that ControlManager::switchController blocks ros service process at [there](https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager/src/controller_manager.cpp#L506).

The bug hangs the process of the node with ControllerManager and CPU usage goes to 98 - 100 %.
There are some reasons why this problem was occurred:

1. ros::shutdown() only turns off ros::ok(), if and only if the internal step of the function reaches at [the code](https://github.com/ros/ros_comm/blob/melodic-devel/clients/roscpp/src/libros/init.cpp#L611).
2. even if ControllerManager::switchController() sees ros::ok() flag in [its loop](https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager/src/controller_manager.cpp#L507), the flag never goes to be turned off, because ControllerManager::switchController() holds ros service, and ros service manager cannot shutdown at [there](https://github.com/ros/ros_comm/blob/melodic-devel/clients/roscpp/src/libros/init.cpp#L604) in ros::shutdown() (just before [turning off g_ok of ros::ok()](https://github.com/ros/ros_comm/blob/melodic-devel/clients/roscpp/src/libros/init.cpp#L611)).
3. when the node with ControllerManager was killed by Ctrl+C, if the external node (controller_manager or spawner) calls switchController and waits for [**please_switch_**](https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager/src/controller_manager.cpp#L506), ControllerManager::switchController blocks forever.
4. but unfortunatelly, ControllerManager has [no destruction mechanisms](https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager/src/controller_manager.cpp#L68), it cannot guess that it was in shutdown process or not, when ros::ok() never turns into false (see above 1).
5. please_switch_ variable will turn off, only if user calls ControllerManager::upate() explicitly. but however, it's difficult for him/her to see switchController is calling in its destruction process. 
6. additionally to say, [usleep(100)](https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager/src/controller_manager.cpp#L509) is really too short duration to sleep. that may cause "spin-lock sleep" in the Linux kernel. that may be occurred below 200 (us) of sleeping. this makes CPU usage up to nearly 100%.

Then, I suggest 4 options:

1. explicitly mention that in ROS Wiki: **not to invoke a node with ControllerManger and a node controller_manger or spawner in a single roslaunch file**
2. or switchController() holds ros service in finite duration.
`while( ros::ok() && please_switch_ && (limit_time > ros::Time::now()) )`
  and also controller_manager and spawner need retry mechanism to switch controller when the service was failed.
3. use a little bit longer sleep. such that `usleep(500)`.
4. ControllerManager has destruction mechanism, such that (but sorry, it's not perfect):
`ControllerManager::~ControllerManager()
{
 is_shutdown = true;
 boost::recursive_mutex::scoped_lock guard(controllers_lock_);  /* join */
}`
`ControllerManager::switchController()
{ ... while( ros::ok() && please_switch_ && !is_shutdown ) ...
}`

yours sincerely.

ipa-mdl, 2018-10-19:
I would suggest that you should call `update` in a (real-time) thread that is not bound to ROS.
Especially, do not use a ROS timer to trigger it. (lesson learned..)

`switchController` is the non-RT/ROS side, `update` is the RT-part.

canalteks, 2018-10-19:
@ipa-mdl , thank you for your comment.

But sorry, to see my comment more precisely.

This lock-up problem does not concern about non-RT or ROS.
`ControllerManager::switchController()` function will block process of ROS service, when `switchControllerSrv` call and `Ctrl+C` event are simultaneously occurred (they are non-RT/ROS events).
But the blocking could not be released until calling `ControllerManager::update` (RT-part).

ipa-mdl mentioned that switchController is non-RT/ROS side, and update is RT-part. Yes, I know your opinion is right. So that, this lock-up can happen.

closed 2018-10-19 by canalteks