ros-controls/ros_control #366: Control multiple joints at the same time

innovatim, 2018-10-20:
For example, there is a robot model with two controllers, here is the yaml file of the controls:
```
myrobot:
  # Publish all joint states -----------------------------------
  joint_state_controller:
    type: joint_state_controller/JointStateController
    publish_rate: 50  
  
  # Position Controllers ---------------------------------------
  controller_1:
    type: effort_controllers/JointPositionController
    joint: joint_1
    pid: {p: 10.0, i: 1.0, d: 0.01}

  #Velocity Controlle
  controller_2:
    type: effort_controllers/JointVelocityController
    joint: joint_2
    pid: {p: 100.0, i: 10.0, d: 0.01}
```
 Is there any way to control both controllers at the same time? It is best to use the gui tool. Any opinion will be appreciated.

bmagyar, 2018-10-20:
If you are looking for a controller that does both, there isn't one at the moment but I don't think it's impossible. In fact one could easily define something along the lines of the JointXYZGroupController patterns where the interface used for each joint is defined in the configuration file. The underlying PID class would be the same.

What do you mean by gui tool? 

If it is only to send commands to the controllers at the same time and you don't have hard time-constraints, just publish them from the same script.

innovatim, 2018-10-20:
@bmagyar   As you said, my idea is to merely send commands to the controllers. As for gui tool, that's because I wanna control a car-like robot, so it's best to have a gui tool(just like [rqt_robot_steering](http://wiki.ros.org/rqt_robot_steering)) to publish messages to the controllers' corresponding topics at any time.

[image]

bmagyar, 2018-10-22:
If you are happy with the rqt_robot_steering plugin, I'd modify it or make up another node that converts the twist message into steering + velocity commands following the car's kinematics/dynamics.

As this is not highly out of scope for *github issues*, I recommend taking this conversation to [ROS Answers](https://answers.ros.org/questions/). We are using the issue tracker for bugs and possible improvements, not for support. Thank you.

closed 2018-10-22 by bmagyar