ros-controls/ros_control #367: Can't locate node [spawner] in package [controller_manager]

innovatim, 2018-10-21:
Hello, everyone. When I execute `roslaunch rrbot_control rrbot_control.launch` from [rrbot](http://gazebosim.org/tutorials/?tut=ros_control), the terminal will yield `ERROR: cannot launch node of type [controller_manager/spawner]: can't locate node [spawner] in package [controller_manager]`, but I can't fix this error by following the [instructions](https://answers.ros.org/question/298010/error-cannot-launch-node-of-type-controller_managerspawner-cant-locate-node-spawner-in-package-controller_manager/#298038) strictly. Oddly, this error wouldn't appear a few days ago, and I have not made changes to the system.

bmagyar, 2018-10-21:
Have your copied the workspace around? Depending on the different file
systems stuff travels through, the executable flags may be dropped. Try
`chmod +x` on the spawner.

innovatim, 2018-10-21:
How can I get the location of `spawner` ?

bmagyar, 2018-10-22:
On installation things go into different folders, not only where roscd takes you. I thought you had a source-based installation as that's pretty much the only case when (still not normally) you may encounter this. If this happens with your package-installed version, you probably fiddled around where you shouldn't have, try reinstalling the package.