ros-controls/ros_control #368: starting and stopping and starting again on switch controllers

bomarali, 2018-10-26:
Hi all!

I am writing a custom controller for the Franka robot. The controller inherits from the ControllerBase. I notice that when the controller is created using the controller_manager spawner the controller calls starting then stopping and then starting again. Is it intended to work like this?

miguelprada, 2018-10-29:
I'm not very familiar with the Franka hardware interface implementation, but it may be that what you're seeing is [this update call resetting the controllers](https://github.com/frankaemika/franka_ros/blob/49e5ac1055e332581b4520a1bd9ac8aaf4580fb1/franka_control/src/franka_control_node.cpp#L197). That would explain the stop/start cycle.

bomarali, 2018-10-29:
Just to make sure, do you say that neither the ControllerBase not controller_manager are designed to do the start stop start? Instead what happens is the spawner starts the controller and the franka_control restarts it? Do I understand your idea correctly?

miguelprada, 2018-10-29:
Yes, that's what I _think_ might be happening.

bomarali, 2018-10-29:
Ok, thank you very much!
I think I will try commenting that bit out and see what happens.

bomarali, 2018-10-29:
I didn't comment out the reset, instead I have just added ros_info there to see the execution order. Going by time stamps it can be seen that the reset of the franka_controller happens after start stop of my custom controller. My guess is the franka_control_node makes sure that the robot does not violate joint_limits and is not related to the custom controller start stop issue?
Please do tell if you have any other ideas! Thanks!

AndyZe, 2020-04-20:
I'm very late to respond, but ...

Re. that line that miguelprada referenced:

The _true_ arg here resets the controllers:

`control_manager.update(now, period, true);`

Then the next line might be resetting them again:

`franka_control.reset();`

So it might be a mistake, the controllers are reset one extra time. (Not 100% sure, I don't have a Franka arm to test)