ros-controls/ros_control #372: Publishing Cartesian frame 'data': cartesian state controller?

gavanderhoorn, 2019-01-30:
More a question / request for comments really: how do people here typically expose "cartesian data" from hardware interfaces?

I've seen a few implementations that add `RealtimePublisher`s to the hw iface and publish either transforms or `geometry_msgs/PoseStamped` (or similar messages). While that works, it does seem to violate separation of concerns a bit, as -- as far as I understand `ros_control` -- the hw iface doesn't interface with any ROS infrastructure directly, but essentially only implements a transformation layer. It's also rare for hw ifaces to have subscribers (or at least: I feel it should be rare).

I don't like asymmetry, so for a setup here I've implemented something like a "cartesian frame state controller" (that's its actual name right now :unamused:) that uses a custom interface + handle to claim "frame resources" from hw ifaces and publish them as `tf2_msgs/TFMessage` (either to `/tf` or to `/tf_static`). The hw iface is responsible for converting whatever internal representation it uses for Cartesian data into standard position + quaternion, so the controller merely converts syntax.

Conceptually, this seems similar to how `joint_state_controller` (and f/t and imu sensors) work(s).

I'm curious to know how others have approached this sort of situation (perhaps you find publishing from hw ifaces is actually ok) and whether having such a cartesian frame state controller as part of `ros_controllers` would make sense.

gavanderhoorn, 2019-01-30:
> I reckon you're referring to "native" Cartesian space information provided by the hardware itself (such as the calibrated tool pose in `ur_modern_driver`), as opposed to information derived from the joint space data, which is usually published as `JointState` messages and converted to `tf` messages using `robot_state_publisher` and the robot's geometric model.

yes. 'native' cartesian data.

The `ur_modern_driver` frame `tool0_controller` would be one example.

> I'm not aware of any generic Cartesian Pose/Twist interfaces, but I think it would be a nice addition to `ros_control` to add them in `hardware_interface`. Inspiration could be taken from [`franka_ros`](https://github.com/frankaemika/franka_ros), which includes such a custom interface in [`franka_cartesian_command_interface.h`](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/franka_hw/include/franka_hw/franka_cartesian_command_interface.h)

I've looked at the `franka_ros` implementation and it is quite nice. The command interface is -- as the name implies -- about commanding/controlling a Cartesian pose. (Industrial) robot controllers however sometimes provide Cartesian frame data that would be nice to be able to publish (as TF frames fi), but those are not always controllable (similar to how an IMU sensor is not (position) controllable).

miguelprada, 2019-01-30:
> I've looked at the franka_ros implementation and it is quite nice. The command interface is -- as the name implies -- about commanding/controlling a Cartesian pose. (Industrial) robot controllers [...]

In that case, I think [`franka_state_interface.h`](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/franka_hw/include/franka_hw/franka_state_interface.h) is closer to what you're looking for. This particular interface contains quite a lot of information that's specific to the Panda robot, but could probably be stripped down and generalized.

gavanderhoorn, 2019-01-30:
Thanks for the suggestions :)

I wasn't looking for examples or things I could copy/adapt per se, but more some sort of consensus on what would be an appropriate way to deal with these sort of resources and how to integrate them into a `ros_control` stack.

I'll see if I can generalise what I have and perhaps propose a PR at some point.

gavanderhoorn, 2019-02-03:
I'll close this for now and come back to it when appropriate.

closed 2019-02-03 by gavanderhoorn