ros-controls/ros_control #378: Is an anti-slip controller, suitable as a ros_control plugin?

gstavrinos, 2019-03-14:
Hello everyone,

I am currently studying on how to most effectively implement a robot controller that compensates for wheel slippage, based on encoders (raw odom) and IMU (acceleration and velocity) inputs.

My main concern is: Would a ros_control plugin be suitable for my goal, or should I just implement a normal node that just overrides cmd_vel commands using a twist multiplexer? Keep in mind, that most people will already have a controller for their robot, so using this second controller should work only complementary.

Thanks in advance for your thoughts.

George

LazyEngineerToBe, 2019-12-07:
@gstavrinos Did you find a solution to this issue?

gstavrinos, 2019-12-09:
@LazyEngineerToBe Unfortunately not, but please keep me updated if you have any progress on this. (Nice username btw)