ros-controls/ros_control #383: No controllers found?

GunnarMaer, 2019-07-04:
Hello everyone,

I'm not a hundred percent sure if this is the correct repo, because I work with MoveIt and want to control two robots (Panda by Franka Emika, therefore I'm using franka_ros / franka_control and UR5 by Universal Robots, therefore I'll be using ROS-Industrial). So I could be that my issue lies in the area of MoveIt, franka_ros or ROS-Industrial. If you think so, feel free to correct me.

### My system:

- ROS Melodic
- Ubuntu 18.04.2 LTS
- Intel® Xeon(R) CPU E3-1225 v3 @ 3.20GHz × 4
- Realtime Kernel (4.19.37-rt20-rc1 # 1 SMP PREEMPT RT)

### What works so far: 
I created a xacro-file to combine the URDFs of both robots and created a MoveIt-config package. This package works fine in simulation with RViz. I'm able to give poses for both robots and they are reached. Now I want to move to the real robots and started with the Panda robot only. I tried the example_controllers and the connection to the robot looked good. I created a launch file with loads the controllers.yaml. When I open the controller manager in rqt or with the console, my specified controllers are available and I am able to control the joints with the joint  trajectory controller tab in rqt.

- In my launch-file, I basically do these steps:
- launch the planning-context file
- launch the franka_control_node, including its settings like the IP adress
- loading the controllers (franka_control/config/default_controllers.yaml and the usual controllers.yaml)
- setting the moveit_controller_manager
- starting the controller_spawner, including my needed controllers
- start the "controller" which publishes the joint states and the corresponding robot_state_publisher
- starting the move_group node, including all necessary settings
- start RViz

### The problem:
However, when I give a desired pose through MoveIt (i.e. with a self written node [which worked fine with simulation only and RViz] and the move() command or with the MotionPlanning window inside RViz), it seems that no controllers are found:
```
[ERROR] [1561553234.640997333]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1561553234.641057622]: Known controllers and their joints:

[ERROR] [1561553234.641112443]: Apparently trajectory initialization failed
```
It seems as if no controllers were found.. But why can I see them in the controller manager then?
I get the same error message for the UR5, but I guess this is not unusual as I haven't connected it yet. 

So, to be concrete, my questions are:

- Have I missed any steps while changing from simulation to reality?

- How can it be that my controllers are found in the controller manager, but the execution fails warning me that no controllers are known?

Thank you very much for any advice! Any help is highly appreciated.

gavanderhoorn, 2019-07-04:
This is a cross-post of [Setting up controllers with MoveIt: No controllers found](http://answers.ros.org/question/327031) on ROS Answers.

I've posted an answer there.

Summarising: this is (almost certainly) not a `ros_control` related problem, but a MoveIt one.

bmagyar, 2019-07-04:
Thanks @gavanderhoorn , closing this one...

closed 2019-07-04 by bmagyar

qyp-robot, 2021-07-16:
Hello! @gavanderhoorn
When I use moveit and gazebo co-simulation, when I use the position controller, the trajectory planned in moveit, gazebo can execute normally. But when I use the effort controller, the terminal reports an error as follows:

Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ] [ERROR]:Known controllers and their joints:

Of course, when I replaced the effort controller, I completely configured it according to the position controller method. I don’t know where the problem is or whether I need to add other things. When I consulted others before, some people said that moveit is based on location. In the servo system, the position controller can run normally, but the effort controller needs to write a client. Is this the right solution? If so, how do I write the client? He is very important to me!

Looking forward to your answers, thank you very much!

gavanderhoorn, 2021-07-16:
@qyp-robot: you are now cross-posting [Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ] [ERROR]:Known controllers and their joints:](https://answers.ros.org/questions/382628) from ROS Answers.

Similar to the previous poster: please don't cross-post.

qyp-robot, 2021-07-16:
thank you for your reply!@gavanderhoorn
But the link you gave is no longer valid and the display page is missing. Can you provide me with it again? Thank you very much!