ros-controls/ros_control #389: controller_manger doesn't work in Melodic anymore

boilerbots, 2019-10-07:
Something seems to have happened to the controller manager in melodic.

I have 

> apt show ros-melodic-ros-control
Package: ros-melodic-ros-control
Version: 0.15.1-0bionic.20190911.200321
Priority: extra
Section: misc
Maintainer: Bence Magyar <bence.magyar.robotics@gmail.com>
Installed-Size: 14.3 kB
Depends: ros-melodic-combined-robot-hw, ros-melodic-combined-robot-hw-tests, ros-melodic-controller-interface, ros-melodic-controller-manager, ros-melodic-controller-manager-msgs, ros-melodic-controller-manager-tests, ros-melodic-hardware-interface, ros-melodic-joint-limits-interface, ros-melodic-realtime-tools, ros-melodic-transmission-interface
Homepage: http://ros.org/wiki/ros_control
Download-Size: 2,712 B
APT-Sources: http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages

The manager hangs and fails to spawn the controllers. A simple service call to list the controllers no longer works, it hangs and never returns a result.

I noticed this failure last week and it is still bad. Has anyone else noticed this?

bmagyar, 2019-10-07:
Can you try a source build and running all the tests with your setup?

boilerbots, 2019-10-08:
I already tried that. I also reverted the usleep change in the manager too.

bmagyar, 2019-10-08:
@ipa-mdl have you guys seen this?

ipa-mdl, 2019-10-08:
@boilerbots: Did you rebuild your workspace (completely)? What driver are you using?

ipa-mdl, 2019-10-08:
@bmagyar: The list service is fully covered in a [unit test](https://github.com/ros-controls/ros_control/blob/melodic-devel/controller_manager_tests/test/controller_manager_scripts.py).  The release is out for a year now. I doubt that this is a problem with `ros_control`.

boilerbots, 2019-10-08:
I have resolved the problem, sorry that implicate ros_control. It turns out the problem came from a very old change in the ros_control_boilerplate project that seems to not have been pushed out until just recently.

Here is the diff that caused the problem, you see that they removed the Timer creation from the constructor that was calling the update function, our controller wasn't being called and could not respond to the controller service calls to change states which messed up a whole lot of things.

https://github.com/PickNikRobotics/ros_control_boilerplate/pull/20/commits/7199b3cb84534daf9887aa8b8a1f17f48e2e11b7

closed 2019-10-08 by boilerbots